wimleers/homebridge-sma-home-manager #2: Plugin does not load properties

Scream4Life, 2023-02-27:
**Paste of Logs:**
```
Feb 27 10:07:26 raspberrypi homebridge[22712]: [27.2.2023, 10:07:26] ---
Feb 27 10:07:26 raspberrypi homebridge[22712]: [27.2.2023, 10:07:26] The plugin "homebridge-sma-home-manager" requires Node.js version of >=18.2.1 which does not satisfy the current Node.js version of v16.19.1. You may need to upgrade your installation of Node.js - see https://homebridge.io/w/JTKEF
Feb 27 10:07:26 raspberrypi homebridge[22712]: [27.2.2023, 10:07:26] Loaded plugin: homebridge-sma-home-manager@1.0.0
Feb 27 10:07:26 raspberrypi homebridge[22712]: [27.2.2023, 10:07:26] Registering platform 'homebridge-sma-home-manager.SMAHomeManager'
Feb 27 10:07:26 raspberrypi homebridge[22712]: [27.2.2023, 10:07:26] ---

Feb 27 10:07:56 raspberrypi homebridge[22712]: [27.2.2023, 10:07:56] [SMAHomeManager] Initializing SMAHomeManager platform...
Feb 27 10:07:57 raspberrypi homebridge[22712]: [27.2.2023, 10:07:57] Loading 12 accessories...

Feb 27 10:08:11 raspberrypi homebridge[22712]: [27.2.2023, 10:08:11] TypeError: Cannot read properties of undefined (reading 'forEach')
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at SMAHomeManager.accessories (/usr/lib/node_modules/homebridge-sma-home-manager/index.js:282:30)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at /usr/lib/node_modules/homebridge/src/bridgeService.ts:534:24
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at new Promise (<anonymous>)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at BridgeService.loadPlatformAccessories (/usr/lib/node_modules/homebridge/src/bridgeService.ts:528:12)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at /usr/lib/node_modules/homebridge/src/server.ts:479:42
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at Array.forEach (<anonymous>)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at Server.loadPlatforms (/usr/lib/node_modules/homebridge/src/server.ts:401:27)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at Server.start (/usr/lib/node_modules/homebridge/src/server.ts:176:29)
Feb 27 10:08:11 raspberrypi homebridge[22712]: [27.2.2023, 10:08:11] TypeError: Cannot read properties of undefined (reading 'buffer')
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at SMAHomeManager.<anonymous> (/usr/lib/node_modules/homebridge-sma-home-manager/index.js:330:25)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at ModbusRTU.writeFC4 (/usr/lib/node_modules/homebridge-sma-home-manager/node_modules/modbus-serial/index.js:544:19)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at ModbusRTU.writeFC3 (/usr/lib/node_modules/homebridge-sma-home-manager/node_modules/modbus-serial/index.js:530:10)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at ModbusRTU.converted [as readHoldingRegisters] (/usr/lib/node_modules/homebridge-sma-home-manager/node_modules/modbus-serial/apis/promise.js:36:27)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at SMAHomeManager._readInverterMetadata (/usr/lib/node_modules/homebridge-sma-home-manager/index.js:329:11)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at SMAHomeManager.<anonymous> (/usr/lib/node_modules/homebridge-sma-home-manager/index.js:206:8)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at listOnTimeout (node:internal/timers:559:17)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at processTimers (node:internal/timers:502:7)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at SMAHomeManager.<anonymous> (/usr/lib/node_modules/homebridge-sma-home-manager/index.js:330:25)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at ModbusRTU.writeFC4 (/usr/lib/node_modules/homebridge-sma-home-manager/node_modules/modbus-serial/index.js:544:19)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at ModbusRTU.writeFC3 (/usr/lib/node_modules/homebridge-sma-home-manager/node_modules/modbus-serial/index.js:530:10)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at ModbusRTU.converted [as readHoldingRegisters] (/usr/lib/node_modules/homebridge-sma-home-manager/node_modules/modbus-serial/apis/promise.js:36:27)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at SMAHomeManager._readInverterMetadata (/usr/lib/node_modules/homebridge-sma-home-manager/index.js:329:11)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at SMAHomeManager.<anonymous> (/usr/lib/node_modules/homebridge-sma-home-manager/index.js:206:8)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at listOnTimeout (node:internal/timers:559:17)
Feb 27 10:08:11 raspberrypi homebridge[22712]:     at processTimers (node:internal/timers:502:7)
```

**Config:**

```json
{
            "signals": {
                "offGrid": true,
                "noSun": true,
                "highImport": false
            },
            "platform": "SMAHomeManager"
        }

```

**Screenshots:**

**Inverter:**

Home Manager 2 640: Sunny Home Manager 2.0
EVC22-3AC-10 166: SMA EV CHARGER 22
SBS3.7-10 721: Sunny Boy Storage 3.7 (Modbus TCP-Server activated, Port 502)
STP 15000TL-30 481: Sunny Tripower 15000TL-30 (Modbus TCP-Server activated, Port 502)

**Homebridge Environment:**

* **Node.js Version**: 16.19.1
* **NPM Version**: 8.19.3
* **Homebridge Version**: 1.6.0
* **homebridge-sma-inverter Version**: 2.83.3.R
* **Operating System**: Raspbian

Scream4Life, 2023-02-27:
Now bridged with this log (my first bridge - but it doesn't show up a QR)
```
Feb 27 10:34:01 raspberrypi homebridge[27476]: [27.2.2023, 10:34:01] [homebridge-sma-home-manager] Loaded homebridge-sma-home-manager v1.0.0 child bridge successfully
Feb 27 10:34:02 raspberrypi homebridge[27476]: /usr/lib/node_modules/homebridge-sma-home-manager/index.js:282
Feb 27 10:34:02 raspberrypi homebridge[27476]: 		this.signalsConfig.surplus.forEach((signal, index) => {
Feb 27 10:34:02 raspberrypi homebridge[27476]:                              ^
Feb 27 10:34:02 raspberrypi homebridge[27476]: TypeError: Cannot read properties of undefined (reading 'forEach')
Feb 27 10:34:02 raspberrypi homebridge[27476]:     at SMAHomeManager.accessories (/usr/lib/node_modules/homebridge-sma-home-manager/index.js:282:30)
Feb 27 10:34:02 raspberrypi homebridge[27476]:     at /usr/lib/node_modules/homebridge/src/bridgeService.ts:534:24
Feb 27 10:34:02 raspberrypi homebridge[27476]:     at new Promise (<anonymous>)
Feb 27 10:34:02 raspberrypi homebridge[27476]:     at BridgeService.loadPlatformAccessories (/usr/lib/node_modules/homebridge/src/bridgeService.ts:528:12)
Feb 27 10:34:02 raspberrypi homebridge[27476]:     at ChildBridgeFork.startBridge (/usr/lib/node_modules/homebridge/src/childBridgeFork.ts:155:36)
Feb 27 10:34:02 raspberrypi homebridge[27476]: [27.2.2023, 10:34:02] [homebridge-sma-home-manager] Child bridge process ended
Feb 27 10:34:02 raspberrypi homebridge[27476]: [27.2.2023, 10:34:02] [homebridge-sma-home-manager] Process Ended. Code: 1, Signal: null
Feb 27 10:34:09 raspberrypi homebridge[27476]: [27.2.2023, 10:34:09] [homebridge-sma-home-manager] Restarting Process...
Feb 27 10:34:13 raspberrypi homebridge[27476]: [27.2.2023, 10:34:13] [homebridge-sma-home-manager] Launched child bridge with PID 28067

```

Scream4Life, 2023-02-27:
A couple of hours later I'm now on 
Node.js Version: 18.14.2
Operating System: Raspbian GNU/Linux Buster (10)

The warning disappeared but the error are still there

```
Feb 27 14:13:10 raspberrypi homebridge[5114]: [27.2.2023, 14:13:10] ---
Feb 27 14:13:11 raspberrypi homebridge[5114]: [27.2.2023, 14:13:11] Loaded plugin: homebridge-sma-home-manager@1.0.0
Feb 27 14:13:11 raspberrypi homebridge[5114]: [27.2.2023, 14:13:11] Registering platform 'homebridge-sma-home-manager.SMAHomeManager'
Feb 27 14:13:11 raspberrypi homebridge[5114]: [27.2.2023, 14:13:11] ---

Feb 27 14:13:54 raspberrypi homebridge[5114]: [27.2.2023, 14:13:54] [SMAHomeManager] Initializing SMAHomeManager platform...
Feb 27 14:13:54 raspberrypi homebridge[5114]: [27.2.2023, 14:13:54] [SMAHomeManager] Initializing child bridge 0E:EC:86:B6:49:44
Feb 27 14:13:54 raspberrypi homebridge[5114]: [27.2.2023, 14:13:54] Loading 12 accessories...

Feb 27 14:14:15 raspberrypi homebridge[5114]: [27.2.2023, 14:14:15] [homebridge-sma-home-manager] Launched child bridge with PID 5486
Feb 27 14:14:30 raspberrypi homebridge[5114]: /usr/lib/node_modules/homebridge-sma-home-manager/index.js:282
Feb 27 14:14:30 raspberrypi homebridge[5114]: 		this.signalsConfig.surplus.forEach((signal, index) => {
Feb 27 14:14:30 raspberrypi homebridge[5114]:                              ^
Feb 27 14:14:30 raspberrypi homebridge[5114]: TypeError: Cannot read properties of undefined (reading 'forEach')
Feb 27 14:14:30 raspberrypi homebridge[5114]:     at SMAHomeManager.accessories (/usr/lib/node_modules/homebridge-sma-home-manager/index.js:282:30)
Feb 27 14:14:30 raspberrypi homebridge[5114]:     at /usr/lib/node_modules/homebridge/src/bridgeService.ts:534:24
Feb 27 14:14:30 raspberrypi homebridge[5114]:     at new Promise (<anonymous>)
Feb 27 14:14:30 raspberrypi homebridge[5114]:     at BridgeService.loadPlatformAccessories (/usr/lib/node_modules/homebridge/src/bridgeService.ts:528:12)
Feb 27 14:14:30 raspberrypi homebridge[5114]:     at ChildBridgeFork.startBridge (/usr/lib/node_modules/homebridge/src/childBridgeFork.ts:155:36)
Feb 27 14:14:30 raspberrypi homebridge[5114]: [27.2.2023, 14:14:30] [homebridge-sma-home-manager] Child bridge process ended
Feb 27 14:14:30 raspberrypi homebridge[5114]: [27.2.2023, 14:14:30] [homebridge-sma-home-manager] Process Ended. Code: 1, Signal: null
Feb 27 14:14:37 raspberrypi homebridge[5114]: [27.2.2023, 14:14:37] [homebridge-sma-home-manager] Restarting Process...
Feb 27 14:19:13 raspberrypi homebridge[5114]: [27.2.2023, 14:19:13] [homebridge-sma-home-manager] Child bridge process ended
Feb 27 14:19:13 raspberrypi homebridge[5114]: [27.2.2023, 14:19:13] [homebridge-sma-home-manager] Process Ended. Code: 1, Signal: null
Feb 27 14:19:20 raspberrypi homebridge[5114]: [27.2.2023, 14:19:20] [homebridge-sma-home-manager] Restarting Process...
Feb 27 14:19:24 raspberrypi homebridge[5114]: [27.2.2023, 14:19:24] [homebridge-sma-home-manager] Launched child bridge with PID 7136
Feb 27 14:19:25 raspberrypi homebridge[5114]: [2/27/2023, 2:19:25 PM] Registering platform 'homebridge-sma-home-manager.SMAHomeManager'
Feb 27 14:19:25 raspberrypi homebridge[5114]: [27.2.2023, 14:19:25] [homebridge-sma-home-manager] Loaded homebridge-sma-home-manager v1.0.0 child bridge successfully
Feb 27 14:19:26 raspberrypi homebridge[5114]: /usr/lib/node_modules/homebridge-sma-home-manager/index.js:282
Feb 27 14:19:26 raspberrypi homebridge[5114]: 		this.signalsConfig.surplus.forEach((signal, index) => {
Feb 27 14:19:26 raspberrypi homebridge[5114]:                              ^
Feb 27 14:19:26 raspberrypi homebridge[5114]: TypeError: Cannot read properties of undefined (reading 'forEach')
Feb 27 14:19:26 raspberrypi homebridge[5114]:     at SMAHomeManager.accessories (/usr/lib/node_modules/homebridge-sma-home-manager/index.js:282:30)
Feb 27 14:19:26 raspberrypi homebridge[5114]:     at /usr/lib/node_modules/homebridge/src/bridgeService.ts:534:24
Feb 27 14:19:26 raspberrypi homebridge[5114]:     at new Promise (<anonymous>)
Feb 27 14:19:26 raspberrypi homebridge[5114]:     at BridgeService.loadPlatformAccessories (/usr/lib/node_modules/homebridge/src/bridgeService.ts:528:12)
Feb 27 14:19:26 raspberrypi homebridge[5114]:     at ChildBridgeFork.startBridge (/usr/lib/node_modules/homebridge/src/childBridgeFork.ts:155:36)
Feb 27 14:19:26 raspberrypi homebridge[5114]: [27.2.2023, 14:19:26] [homebridge-sma-home-manager] Child bridge process ended
Feb 27 14:19:26 raspberrypi homebridge[5114]: [27.2.2023, 14:19:26] [homebridge-sma-home-manager] Process Ended. Code: 1, Signal: null
Feb 27 14:19:31 raspberrypi sudo[7173]:       pi : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/kill -9 5682
Feb 27 14:19:31 raspberrypi sudo[7173]: pam_unix(sudo:session): session opened for user root by (uid=0)
Feb 27 14:19:31 raspberrypi sudo[7173]: pam_unix(sudo:session): session closed for user root
Feb 27 14:19:32 raspberrypi homebridge[5114]: [27.2.2023, 14:19:32] [Meross] [Hub Strom] has been reported [online].
Feb 27 14:19:32 raspberrypi sudo[7186]:       pi : TTY=pts/1 ; PWD=/var/homebridge ; USER=root ; COMMAND=/bin/journalctl -o cat -n 500 -f -u homebridge
Feb 27 14:19:33 raspberrypi sudo[7186]: pam_unix(sudo:session): session opened for user root by (uid=0)
Feb 27 14:19:33 raspberrypi homebridge[5114]: [27.2.2023, 14:19:33] [homebridge-sma-home-manager] Restarting Process...

```

Scream4Life, 2023-03-01:
now...inserted a test surplus signal
```
{
            "signals": {
                "offGrid": true,
                "noSun": true,
                "highImport": false
            },
            "surplusSignals": [
                {
                    "label": "Test",
                    "watts": 10,
                    "minutes": 10
                }
            ],
```

```
[1.3.2023, 15:29:14] [homebridge-sma-home-manager] Loaded homebridge-sma-home-manager v1.0.0 child bridge successfully
/usr/lib/node_modules/homebridge-sma-home-manager/index.js:330
                                serialNumber = data.buffer.readUInt32BE();
                        ^
TypeError: Cannot read properties of undefined (reading 'buffer')
    at SMAHomeManager.<anonymous> (/usr/lib/node_modules/homebridge-sma-home-manager/index.js:330:25)
    at ModbusRTU.writeFC4 (/usr/lib/node_modules/homebridge-sma-home-manager/node_modules/modbus-serial/index.js:544:19)
    at ModbusRTU.writeFC3 (/usr/lib/node_modules/homebridge-sma-home-manager/node_modules/modbus-serial/index.js:530:10)
    at ModbusRTU.converted [as readHoldingRegisters] (/usr/lib/node_modules/homebridge-sma-home-manager/node_modules/modbus-serial/apis/promise.js:36:27)
    at SMAHomeManager._readInverterMetadata (/usr/lib/node_modules/homebridge-sma-home-manager/index.js:329:11)
    at SMAHomeManager.<anonymous> (/usr/lib/node_modules/homebridge-sma-home-manager/index.js:206:8)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)
[1.3.2023, 15:29:15] [homebridge-sma-home-manager] Child bridge process ended
[1.3.2023, 15:29:15] [homebridge-sma-home-manager] Process Ended. Code: 1, Signal: null
[1.3.2023, 15:29:22] [homebridge-sma-home-manager] Restarting Process...
[1.3.2023, 15:29:28] [homebridge-sma-home-manager] Launched child bridge with PID 5328
[3/1/2023, 3:29:28 PM] Registering platform 'homebridge-sma-home-manager.SMAHomeManager'```

wimleers, 2023-03-01:
```
Feb 27 10:08:11 raspberrypi homebridge[22712]: [27.2.2023, 10:08:11] TypeError: Cannot read properties of undefined (reading 'forEach')
```
+
```
Feb 27 10:34:02 raspberrypi homebridge[27476]: TypeError: Cannot read properties of undefined (reading 'forEach')
```

Already fixed 2 days ago: https://github.com/wimleers/homebridge-sma-home-manager/commit/7d60e149dd7892e661f382ac2c8481084657264a — sorry about that! 😅 I've been testing with all functionality enabled, including the optional configuration, and that's how I missed this!

However, this one:
```
[1.3.2023, 15:29:14] [homebridge-sma-home-manager] Loaded homebridge-sma-home-manager v1.0.0 child bridge successfully
/usr/lib/node_modules/homebridge-sma-home-manager/index.js:330
                                serialNumber = data.buffer.readUInt32BE();
                        ^
TypeError: Cannot read properties of undefined (reading 'buffer')
```
looks to be happening because your SMA inverter uses ModBus differently. Will investigate…

wimleers, 2023-03-01:
Hm, that's odd. According to the `SMA_Modbus-TI-en-23` ModBus spec spreadsheet, your inverter ("Sunny Tripower: STP 15000TL-30, STP 20000TL-30, STP 25000TL-30")'s serial number should be in register `30057`:
```
30057 | Serial number |   | U32 | RAW | RO
```
which is identical to what's listed for mine ("Sunny Boy: SB3.0-1AV-40 / SB3.6-1AV-40 / SB4.0-1AV-40 / SB5.0-1AV-40"):
```
30057 | Serial number |   | U32 | RAW | RO
```
… so something else must be boing on here.

## Question 1: does it connect?

Could you please go to `Homebridge settings` and enable `Homebridge Debug Mode`? 🙏 That will print more useful output. In particular, I'm wondering what output you'll see from this:
```
	_connectToInverter: function() {
		this.log.debug("Connecting to inverter.");
		try {
			client.connectTCP(this.inverterAddress);
			client.setID(SMA_MODBUS_CLIENT_ID);
			this.log.debug("Successfully connected to inverter.");
		}
		catch(err) {
			this.log.error("Failed to connect to inverter.", err);
			return;
		}
	},
```
… because it kind of sounds like that's not quite working.

## Question 2: does `169.254.12.3` resolve to your battery inverter or your solar panel inverter?

Another possible cause: you have a Sunny Boy Storage _too_, also with ModBus enabled. I wonder if it's connecting to that instead? If you browse to `169.254.12.3` on your local network, which inverter do you see?

wimleers, 2023-03-01:
Just published https://github.com/wimleers/homebridge-sma-home-manager/releases/tag/1.0.1, which should allow you to remove the "test" PV surplus signal you had set up! 👍

Scream4Life, 2023-03-01:
First of all, thank you for your time and the investigation here. The update has been installed and for now my surplus signal test has been deleted. 

Regarding 2) 169.254.12.3 can‘t be resolved successfully in my network. That seems to be the main problem for now. What is this IP about? My network devices start with 192.168….. Should the inverter open here a new network? Is this some kind of service I need to activate?

wimleers, 2023-03-02:
See https://manuals.sma.de/SBSxx-10/en-US/1685190283.html.

But … perhaps SMA has mysteriously chosen to only make this work for wired connections, not for wireless connections? 🤔 That'd be completely absurd but not impossible … 😬 If so, I'll need to figure out an alternative automatic discovery.

Does `ping SMA[serial number].local` (e.g. `ping SMA0123456789.local`) work? (That may be an alternative based on https://manuals.sma.de/SBSxx-10/en-US/3389030027.html)

What is the output you see after enabling Homebridge's debug mode?

sweh, 2023-03-02:
Hi @wimleers,

I have the same problem with the wrong ip address. The SMA[serial number].local seems to work:

```
$ ping SMA3009870204.local
PING SMA3009870204.local (10.0.1.122) 56(84) bytes of data.
64 bytes from SMA-1285097092.wehrmann.local (10.0.1.122): icmp_seq=1 ttl=64 time=0.438 ms
64 bytes from SMA-1285097092.wehrmann.local (10.0.1.122): icmp_seq=2 ttl=64 time=0.460 ms
```

Could you make the IP oder the serial configurable?

Cheers,
  Sebastian

Scream4Life, 2023-03-02:
Hi, the Sunny Tripower is connected via (switch) Ethernet, not via WiFi. Nevertheless I can't ping it from my computer or raspberry
```
pi@raspberrypi:~ $ sudo ping SMA1980452481.local
ping: SMA1980452481.local: Name or service not known
```

The Sunny Boy Storage inverter and the Sunny Home Manager are reachable via ping.

Scream4Life, 2023-03-02:
I reconsidered my statement about the networking of the inverter. I don't see it listed in my network devices on my DHCP server. I just see the IPs and labels for the Homemanager, EV Charger and Storage inverter. That's going to be tricky than. I don't see a QR code for the Wifi etc and I can't access the LAN ports directly. 

Your plugin can not look only on the Homemanager data, right? Maybe this could be an option to get at least this part up and running.

wimleers, 2023-03-02:
Aha! So then everything is sane and understandable again! 😄

Do I understand correctly then that your PV inverter is not connected to the network at all?

That is very odd and not according to the installation instructions for the SHM (SMA Home Manager) to my knowledge.

I'd really rather avoid supporting a special "production not known" mode, because that not only makes things more complicated, it also makes some things impossible: the SHM only knows *net* flows of power, meaning that without the PV inverter's production information it is impossible to say how much is *actually* consumed or produced:

```
const measurement = {
			timestamp: timestampFromEnergyManager,
			import: importWatts,
			export: exportWatts,
			production: producedWattsFromInverter,
			consumption: importWatts + producedWattsFromInverter - exportWatts,
		};
```

wimleers, 2023-03-02:
You did say:

> STP 15000TL-30 481: Sunny Tripower 15000TL-30 (Modbus TCP-Server activated, Port 502)

How do you know that to be the case if it is not connected?

Scream4Life, 2023-03-02:
It sounds really weird but when I compare it with my DHCP server this seems to be the only explanation. I see all the datapoints in the SMA app, the web portal and I had access to the inverter via SMA Explorer on a PC. Not sure whether the connection is done only via Homemanager. I will investigate this further or try to bring it into the network by my own.

wimleers, 2023-03-02:
That _is_ weird! 😮 

Do you see something like this at https://www.sunnyportal.com/Templates/DeviceProperties.aspx?
[image]

Scream4Life, 2023-03-02:
Yes. I see all the devices on this site.

I think I found it‘s IP address via Sunny Explorer and I can ping it. But not the SmaSerial.local thing. Is it an idea to put these into variables in the config?

Scream4Life, 2023-03-02:
I changed the IP in the index.js and am now see an operational child bridge with constant update every second. Does this look correct to you? By the way…the sun went down hours ago. 

https://user-images.githubusercontent.com/8568430/222573478-720330a4-84d1-4ddc-97fb-7f7908b95015.MOV

wimleers, 2023-03-02:
I really want to avoid address configuration.

Can you install https://apps.apple.com/be/app/discovery-dns-sd-browser/id305441017?l=nl, tap on `_http._tcp` and see if there’s SMA stuff there?

Scream4Life, 2023-03-03:
It shows just the Homemanager and the Battery-Inverter :-(

Scream4Life, 2023-03-03:
By the way, is it possible to extract the data to shortcuts? For some reason I can't see the devices or even the Energy room in the list of HomeKit devices there. From there some kind of reporting could be done.

I'm wondering how you use the data in your daily live once you have it in HomeKit. Automations based on the true/false values on the surplus triggers?

wimleers, 2023-03-03:
> It shows just the Homemanager and the Battery-Inverter :-(

That is *so* weird! 😳

> By the way, is it possible to extract the data to shortcuts? For some reason I can't see the devices or even the Energy room in the list of HomeKit devices there. From there some kind of reporting could be done.

No idea, I never use Shortcuts because the UX is horrible. It’s possible you can’t see them there because these are not standard accessories (impossible to do because HomeKit does not support them).

What kind of reporting are you thinking of?

> I'm wondering how you use the data in your daily live once you have it in HomeKit. Automations based on the true/false values on the surplus triggers?

Absolutely! I have many of these kinds of automations 😊 Been using them for months. I’m writing a blog, it will be published soon.

Use Eve, Home+ or Controller to set these up!

Scream4Life, 2023-03-03:
I have some summaries build in shortcuts which e.g. shows the temperatures of all rooms. Would be nice to add energy data here in order to see whether a room can be headed or cooled down - for free. 
Or with shortcuts you can start asking Siri things about the energy status. But probably she would give wrong answers or put the lights out :-/

wimleers, 2023-03-03:
> I have some summaries build in shortcuts which e.g. shows the temperatures of all rooms.

Sounds interesting! Can you share a screenshot of what the output looks like?

> Would be nice to add energy data here in order to see whether a room can be heated up or cooled down - for free.

I have automations for that too! I just compare outside temperature (a certain range) with inside temperature (another range) and then check if there's a certain level of solar electricity production (which correlates with solar irradiance) and then open my roof windows' shutters 🤓 

> Or with shortcuts you can start asking Siri things about the energy status. But probably she would give wrong answers or put the lights out :-/

There is no way to make that work until Apple adds support for characteristics that relate to energy consumption — https://developer.apple.com/documentation/homekit/hmcharacteristic/characteristic_types#3080269 contains none of that, only:
> Power and switches
> - `HMCharacteristicTypeBatteryLevel`: The battery level of the accessory.
> - `HMCharacteristicTypeChargingState`: The charging state of a battery.
> - `HMCharacteristicTypeContactState`: The state of a contact sensor.
> - `HMCharacteristicTypeOutletInUse`: The state of an outlet.
> - `HMCharacteristicTypePowerState`: The power state of the accessory.
> - `HMCharacteristicTypeStatusLowBattery`: A low battery indicator.
> - `HMCharacteristicTypeOutputState`: The output state of a programmable switch.
> - `HMCharacteristicTypeInputEvent`: The input event of a programmable switch.

sweh, 2023-03-03:
> My example temperature shortcut output (from summer obviously - the pool is a lot cooler now)

That's pretty impressive. Can you share you shortcut with us?

Scream4Life, 2023-03-03:
These shortcuts are working very individual for each home. Mine is basically a query of home devices of the current temperature, defining variables and compare one with each other with some 'if this than that' formulas. In the end I output everything as html text (which is the only programming language that I can write) and pdf overviews. Readable on all iOS devices (even the watch) and Siri can read those things loud (not this exactly...but those outputs) 

The UI is horrible as wimleers wrote, but putting the things together is easy. 

https://www.icloud.com/shortcuts/5994528b76da44d2a2f4efab155e41e0

Scream4Life, 2023-03-16:
> 

> Absolutely! I have many of these kinds of automations 😊 Been using them for months. I’m writing a blog, it will be published soon.

Any progress on your blog, how you make the usage of the data as effective as possible?

wimleers, 2023-03-16:
Soon. I wanted to get #3 fixed first, to avoid more people running into the same problems you ran into!