wimleers/homebridge-sma-home-manager #4: Valid datagram found, but with unknown structure. Discarding. <Buffer 60 81>

sweh, 2023-03-03:
Hi,

I managed to install this nice plugin, but it does not show any data. In the logs I get the following error:

```
[3/3/2023, 9:21:14 AM] [homebridge-sma-home-manager] Valid datagram found, but with unknown structure. Discarding. <Buffer 60 81>
```

Can you assist me on debugging this? Is there a way I can log the buffer in a readable form? I can adjust the code in `index.js` if you tell me what and where.

Thanks in advance,
  Sebastian

sweh, 2023-03-03:
I managed to log the `msg` variable in the logs, but it's not very readable. Looks like an encoding issue?

```
[3/3/2023, 10:01:14 AM] [homebridge-sma-home-manager] Valid datagram found, but with unknown structure. Discarding. <Buffer 60 81>
[3/3/2023, 10:01:14 AM] [homebridge-sma-home-manager] Datagram: SMA�
                                                                    `�������
[3/3/2023, 10:01:20 AM] [homebridge-sma-home-manager] Datagram: SMA�&`e �������t|�f���
[3/3/2023, 10:01:51 AM] [homebridge-sma-home-manager] Datagram: SMA�&`e �������t|�f��
[3/3/2023, 10:02:14 AM] [homebridge-sma-home-manager] Valid datagram found, but with unknown structure. Discarding. <Buffer 60 81>
[3/3/2023, 10:02:14 AM] [homebridge-sma-home-manager] Datagram: SMA�
                                                                    `�������
[3/3/2023, 10:02:22 AM] [homebridge-sma-home-manager] Datagram: SMA�&`e �������t|�f�t�
[3/3/2023, 10:02:53 AM] [homebridge-sma-home-manager] Datagram: SMA�&`e �������t|�f�Һ
[3/3/2023, 10:03:14 AM] [homebridge-sma-home-manager] Valid datagram found, but with unknown structure. Discarding. <Buffer 60 81>
[3/3/2023, 10:03:14 AM] [homebridge-sma-home-manager] Datagram: SMA�
                                                                    `�������
[3/3/2023, 10:03:24 AM] [homebridge-sma-home-manager] Datagram: SMA�&`e �������t|�f�.�
[3/3/2023, 10:03:55 AM] [homebridge-sma-home-manager] Datagram: SMA�&`e �������t|�f���
[3/3/2023, 10:04:14 AM] [homebridge-sma-home-manager] Valid datagram found, but with unknown structure. Discarding. <Buffer 60 81>
[3/3/2023, 10:04:14 AM] [homebridge-sma-home-manager] Datagram: SMA�
                                                                    `�������
[3/3/2023, 10:04:26 AM] [homebridge-sma-home-manager] Datagram: SMA�&`e �������t|�f���
[3/3/2023, 10:04:57 AM] [homebridge-sma-home-manager] Datagram: SMA�&`e �������t|�f�L�
[3/3/2023, 10:05:14 AM] [homebridge-sma-home-manager] Valid datagram found, but with unknown structure. Discarding. <Buffer 60 81>
[3/3/2023, 10:05:14 AM] [homebridge-sma-home-manager] Datagram: SMA�
                                                                    `�������
[3/3/2023, 10:05:28 AM] [homebridge-sma-home-manager] Datagram: SMA�&`e �������t|�f���
[3/3/2023, 10:05:59 AM] [homebridge-sma-home-manager] Datagram: SMA�&`e �������t|�f��
[3/3/2023, 10:06:14 AM] [homebridge-sma-home-manager] Valid datagram found, but with unknown structure. Discarding. <Buffer 60 81>
[3/3/2023, 10:06:14 AM] [homebridge-sma-home-manager] Datagram: SMA�
                                                                    `�������
[3/3/2023, 10:06:30 AM] [homebridge-sma-home-manager] Datagram: SMA�&`e �������t|�f�w�
[3/3/2023, 10:07:01 AM] [homebridge-sma-home-manager] Datagram: SMA�&`e �������t|�f�ؽ
[3/3/2023, 10:07:14 AM] [homebridge-sma-home-manager] Valid datagram found, but with unknown structure. Discarding. <Buffer 60 81>
[3/3/2023, 10:07:14 AM] [homebridge-sma-home-manager] Datagram: SMA�
                                                                    `�������
[3/3/2023, 10:07:32 AM] [homebridge-sma-home-manager] Datagram: SMA�&`e �������t|�f�7�
[3/3/2023, 10:08:03 AM] [homebridge-sma-home-manager] Datagram: SMA�&`e �������t|�f���
[3/3/2023, 10:08:14 AM] [homebridge-sma-home-manager] Valid datagram found, but with unknown structure. Discarding. <Buffer 60 81>
[3/3/2023, 10:08:14 AM] [homebridge-sma-home-manager] Datagram: SMA�
                                                                    `�������
[3/3/2023, 10:08:34 AM] [homebridge-sma-home-manager] Datagram: SMA�&`e �������t|�f��
[3/3/2023, 10:09:05 AM] [homebridge-sma-home-manager] Datagram: SMA�&`e �������t|�f�Q�
[3/3/2023, 10:09:14 AM] [homebridge-sma-home-manager] Valid datagram found, but with unknown structure. Discarding. <Buffer 60 81>
[3/3/2023, 10:09:14 AM] [homebridge-sma-home-manager] Datagram: SMA�
                                                                    `�������
```

sweh, 2023-03-03:
Ok, I guess I was on the wrong track here. Through intensive logging I was able to find out that the required values are read out by the HomeManager. For example, I see `t.production` increasing every few seconds. Still I don't see them in the Home app, it says "This device is currently not supported by the Home app" for all 6 devices.

Any idea what I'm doing wrong?

wimleers, 2023-03-03:
> Ok, I guess I was on the wrong track here. Through intensive logging I was able to find out that the required values are read out by the HomeManager. For example, I see `t.production` increasing [...]

The Home app does not support custom accessories, services and characteristics. Use Home+, Eve or Controller 😊

sweh, 2023-03-03:
Oh I see. With Controller it works for me. Thanks!!!

wimleers, 2023-03-06:
I'm now also getting `Valid datagram found, but with unknown structure. Discarding. <Buffer 60 81>`!

This appears to have been triggered by [my inverter](https://www.sma.de/en/products/solarinverters/sunny-boy-30-36-40-50-60) upgrading to firmware 04.00.75.R. Any chance the same is true for you?

For me too this is happening exactly once every 60 s.

Scream4Life, 2023-03-07:
> I'm now also getting `Valid datagram found, but with unknown structure. Discarding. <Buffer 60 81>`!
> 
> I'm now also getting `Valid datagram found, but with unknown structure. Discarding. <Buffer 60 81>`!
> 
> For me too this is happening exactly once every 60 s.

I‘m getting the same error, but didn‘t touch my inverter firmware. But figures are still rolling in.

wimleers, 2023-03-07:
> I‘m getting the same error, but didn‘t touch my inverter firmware. But figures are still rolling in.

I didn’t touch it either. It was updated automatically on March 4!

TheRealGros, 2023-03-13:
Same issue here. Just installed the plugin.

wimleers, 2023-03-14:
Firmware `4.00.75.R` for Sunny Boy inverters was released on 14.02.2023, and if you [download it from the SMA website](https://files.sma.de/downloads/FW4-00-75-R_SB60-1AV-41.zip), you'll find a 
[READ_ME.pdf](https://github.com/wimleers/homebridge-sma-home-manager/files/10972749/READ_ME.pdf) included.

That says:
```
The update file contains all features of former firmware versions New features and solved issues:
• Unicast protocol for reliable communication between inverter and meter
• Solved communication issue (all LEDs off) after rebooting the inverter
```
… but that protocol has not been documented! 😬 It's unclear whether the datagrams we receive exist solely for discovery (since it's multicast, not unicast) to then set up a direct connection between inverter & SHM, or whether it contains meaningful data.

But we can't use it anyway for this plugin because the update frequency would be too low.

At least we have a very strong clue in the README!

wimleers, 2023-03-14:
Temporary debug output I put in place:
```
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] Valid datagram found, but for inverter communication discovery. Ignoring.
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 0, 83 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 1, 77 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 2, 65 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 3, 0 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 4, 0 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 5, 4 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 6, 2 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 7, 160 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 8, 0 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 9, 0 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 10, 0 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 11, 1 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 12, 0 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 13, 12 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 14, 0 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 15, 16 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 16, 96 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 17, 129 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 18, 0 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 19, 1 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 20, 1 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 21, 153 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 22, 179 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 23, 119 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 24, 51 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 25, 53 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 26, 255 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 27, 255 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 28, 0 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 29, 0 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 30, 0 ]
[3/14/2023, 8:05:02 PM] [homebridge-sma-home-manager] [ 31, 0 ]
…
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] Valid datagram found, but for inverter communication discovery. Ignoring.
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 0, 83 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 1, 77 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 2, 65 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 3, 0 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 4, 0 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 5, 4 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 6, 2 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 7, 160 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 8, 0 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 9, 0 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 10, 0 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 11, 1 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 12, 0 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 13, 12 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 14, 0 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 15, 16 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 16, 96 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 17, 129 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 18, 0 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 19, 1 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 20, 1 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 21, 153 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 22, 179 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 23, 119 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 24, 51 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 25, 53 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 26, 255 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 27, 255 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 28, 0 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 29, 0 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 30, 0 ]
[3/14/2023, 8:06:02 PM] [homebridge-sma-home-manager] [ 31, 0 ]
```

closed 2023-03-15 by wimleers

wimleers, 2023-03-15:
🚢 Fix shipped in https://github.com/wimleers/homebridge-sma-home-manager/releases/tag/1.0.2

commit 12b6392c (2023-03-15): Since inverter firmware 4.00.75.R: "Unicast protocol for reliable communication between inverter and meter", introduced additional Speedwire datagrams to ignore. Fixes #4.