wimleers/homebridge-sma-home-manager #5: Battery Inverter Status

Scream4Life, 2023-03-03:
Wouldn‘t this SMA plugin be the greatest of all time (i mean - it is already!) if also the Battery Inverter Status could be extracted?! 😉 

In case you need a test user 👋

wimleers, 2023-03-03:
Agreed! 😄

Sadly I do not have a battery nor an inverter 😅

1. Is the representation/data aggregation I have created for everything besides a battery making sense to you?
2. How do you propose to integrate it? (I have thought about it while writing this but would love your unbiased input!)

sweh, 2023-03-03:
I have a battery here and a Python plugin that's working. Maybe those hardcoded serials can help you as a starting point?

```
['30057', 'U32', 'RAW', 'serial', ''],
['30201', 'U32', 'ENUM', 'Status',''],
['30051', 'U32', 'ENUM', 'DeviceClass',''],
['30053', 'U32', 'ENUM', 'DeviceID',''],
['30775', 'S32', 'FIX0', 'AC Power', 'W'],
['30517', 'U64', 'FIX3', 'daily yield', 'kWh'],
['30953', 'S32', 'FIX1', 'Device_Temperature', u'\xb0C'],
['30849', 'S32', 'FIX1', 'BatteryTemp', u'\xb0C'],
['30843', 'S32', 'FIX3', 'BatteryAmp', 'A'],
['30851', 'U32', 'FIX2', 'BatteryVolt', 'V'],
['30845', 'U32', 'FIX0', 'BatteryCharge', u'\u0025'],
['30955', 'U32', 'ENUM', 'BatteryState', ''],
['31391', 'U32', 'ENUM', 'BatteryHealth', ''],
['30813', 'S32', 'FIX0', 'AC apparent power', 'VA'],
['30803', 'U32', 'FIX2', 'Grid_Frequency', 'Hz'],
['30513', 'U64', 'FIX3', 'Total_Yield', 'kWh'],
['30521', 'U64', 'FIX0', 'Operating_Time', 's'],
['30525', 'U64', 'FIX0', 'Feed-in_Time', 's']
```

I think that reads: `register`, `size`, `datatype`, `kind`, `unit`. I saw some of that information in your source code, so I suspect that is familiar to you?!

Scream4Life, 2023-03-03:
1. I have to spend more time with it in order to get used to the many output properties that are constantly flickering 😉 But in the end I can find nearly the same figures in the SMA portal as well. 
2. I would suggest another row/device for the energy room displaying "Battery status". Properties should definitely be the charge level in percentage. Further the properties Input Watt and Output Watt could indicate whether energy goes in or out and how much.

wimleers, 2023-03-26:
The investigations in #9 have led me to conclude (see https://github.com/wimleers/homebridge-sma-home-manager/issues/9#issuecomment-1484068936) that at least in @Scream4Life's setup, the PV inverter does not get advertised, only the battery inverter does.

We also know for a fact that this plugin reads data from it just fine, but of course it's ready the flow of energy to/from the battery, rather than only from the solar panels.

I don't know yet how to discover the PV inverter in his case (see https://github.com/wimleers/homebridge-sma-home-manager/issues/9#issuecomment-1474941688), but it seems _somewhat possible_ (although still difficult) to add battery inverter support now, despite my not having one.

wimleers, 2023-03-26:
Using Modbus I can at a minimum verify that we're connected to a solar inverter, thanks to registers:
- `30051`: `8001` = solar inverter, `8007` = battery inverter
- `30053`: a whole range of possible values for the _precise models_ is documented in `SMA_Modbus-TI-en-23.xls`, perhaps I should only allow the ones listed there? 🤔 So far that has not seemed necessary.

wimleers, 2023-03-26:
The device class detection shipped in `1.1.1`, which means that at least the problems that @Scream4Life reported in #9 will now not be possible anymore: the plugin simply won't start anymore.

wimleers, 2023-03-26:
As soon as you update to https://github.com/wimleers/homebridge-sma-home-manager/releases/tag/1.1.1, it will not be possible anymore for this plugin to mistake a battery inverter for a solar inverter: it will refuse to start.

That's a good first step at least 😇

Scream4Life, 2023-03-26:
```
Mar 26 15:50:03 raspberrypi homebridge[20901]: [26.3.2023, 15:50:00] [homebridge-sma-home-manager] Discovered SMA inverter: no
Mar 26 15:50:03 raspberrypi homebridge[20901]: [26.3.2023, 15:50:00] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006124640, FirmwareRevision: '2.11.5.R' }
Mar 26 15:50:03 raspberrypi homebridge[20901]: [26.3.2023, 15:50:01] [homebridge-sma-home-manager] Discovered SMA inverter: {
Mar 26 15:50:03 raspberrypi homebridge[20901]:   DeviceClass: 'battery inverter',
Mar 26 15:50:03 raspberrypi homebridge[20901]:   SerialNumber: 3011693721,
Mar 26 15:50:03 raspberrypi homebridge[20901]:   FirmwareRevision: 'unknown'
Mar 26 15:50:03 raspberrypi homebridge[20901]: }
Mar 26 15:50:03 raspberrypi homebridge[20901]: [26.3.2023, 15:50:01] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006124640, FirmwareRevision: '2.11.5.R' }
Mar 26 15:52:14 raspberrypi homebridge[20901]: [3/26/2023, 3:52:14 PM] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006124640, FirmwareRevision: '2.11.5.R' }
Mar 26 15:52:14 raspberrypi homebridge[20901]: [3/26/2023, 3:52:14 PM] [homebridge-sma-home-manager] The discovered inverter is not a solar inverter! Please create a bug report with as much detail as possible.
Mar 26 15:52:32 raspberrypi homebridge[20901]: [3/26/2023, 3:52:32 PM] [homebridge-sma-home-manager] This plugin is taking long time to load and preventing Homebridge from starting. See https://homebridge.io/w/JtMGR for more info.
```

wimleers, 2023-03-26:
Yep.

If I remember correctly, you used to hack the code to hardcode an IP address for your solar inverter. Do I remember that correctly?

Scream4Life, 2023-03-26:
Probably not. I've inserted it in the _findInverter function now here
[image]

An interesting logging outcome shows the correct serial number and a rather old Firmware probably. 

```
Mar 26 20:43:55 raspberrypi homebridge[10768]: Error: Not running
Mar 26 20:43:55 raspberrypi homebridge[10768]:     at new NodeError (node:internal/errors:399:5)
Mar 26 20:43:55 raspberrypi homebridge[10768]:     at healthCheck (node:dgram:911:11)
Mar 26 20:43:55 raspberrypi homebridge[10768]:     at Socket.dropMembership (node:dgram:859:3)
Mar 26 20:43:55 raspberrypi homebridge[10768]:     at SMAHomeManager._keepMembershipActive (/usr/lib/node_modules/homebridge-sma-home-manager/index.js:802:15)
Mar 26 20:43:55 raspberrypi homebridge[10768]:     at listOnTimeout (node:internal/timers:569:17)
Mar 26 20:43:55 raspberrypi homebridge[10768]:     at processTimers (node:internal/timers:512:7)
Mar 26 20:43:55 raspberrypi homebridge[10768]: [26.3.2023, 20:43:55] [homebridge-sma-home-manager] Child bridge process ended
Mar 26 20:43:55 raspberrypi homebridge[10768]: [26.3.2023, 20:43:55] [homebridge-sma-home-manager] Process Ended. Code: 1, Signal: null
Mar 26 20:44:02 raspberrypi homebridge[10768]: [26.3.2023, 20:44:02] [homebridge-sma-home-manager] Restarting Process...
Mar 26 20:44:07 raspberrypi homebridge[10768]: [26.3.2023, 20:44:07] [homebridge-sma-home-manager] Launched child bridge with PID 11518
Mar 26 20:44:08 raspberrypi homebridge[10768]: [3/26/2023, 8:44:08 PM] Registering platform 'homebridge-sma-home-manager.SMAHomeManager'
Mar 26 20:44:08 raspberrypi homebridge[10768]: [26.3.2023, 20:44:08] [homebridge-sma-home-manager] Loaded homebridge-sma-home-manager v1.1.1 child bridge successfully
Mar 26 20:44:08 raspberrypi homebridge[10768]: [3/26/2023, 8:44:08 PM] Loaded 0 cached accessories from cachedAccessories.0EEC86B64944.
Mar 26 20:44:08 raspberrypi homebridge[10768]: [3/26/2023, 8:44:08 PM] [homebridge-sma-home-manager] Found inverter SMA3011693721 at SMA3011693721.local with IPv4 address 192.168.0.26.
Mar 26 20:44:09 raspberrypi homebridge[10768]: [3/26/2023, 8:44:09 PM] [homebridge-sma-home-manager] Discovered SMA inverter: no
Mar 26 20:44:09 raspberrypi homebridge[10768]: [3/26/2023, 8:44:09 PM] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006124640, FirmwareRevision: '2.11.5.R' }
Mar 26 20:44:10 raspberrypi homebridge[10768]: [3/26/2023, 8:44:10 PM] [homebridge-sma-home-manager] Discovered SMA inverter: { SerialNumber: 1980452352, FirmwareRevision: '0.19.0.N' }
Mar 26 20:44:10 raspberrypi homebridge[10768]: [3/26/2023, 8:44:10 PM] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006124640, FirmwareRevision: '2.11.5.R' }
Mar 26 20:44:10 raspberrypi homebridge[10768]: [3/26/2023, 8:44:10 PM] [homebridge-sma-home-manager] The discovered inverter is not a solar inverter! Please create a bug report with as much detail as possible.
Mar 26 20:44:28 raspberrypi homebridge[10768]: [3/26/2023, 8:44:28 PM] [homebridge-sma-home-manager] This plugin is taking long time to load and preventing Homebridge from starting. See https://homebridge.io/w/JtMGR for more info.

```

The data in my energy room is still garbage with that but I didn't expect something useful here when the "sun is off" 😉

Scream4Life, 2023-03-27:
Hm, that wasn't the right spot either. There is not data coming in. Where do I need to insert the IP of the inverter now?

wimleers, 2023-03-27:
Change it to:
```
_findInverter: function(callback) {
  this.inverterAddress = '192.168.x.y';
  callback();
},
```

@Scream4Life Any chance you would be willing to grant me remote access to your machine (using your mac's "Screen sharing" service in System Settings) so I can try to figure out some way to automatically discover your PV inverter based on the information in mDNS/DNS-SD, your SMA Home Manager or your battery inverter? There _must_ be some way to figure this out automatically!

@sweh Do you also have the problem that @Scream4Life is seeing, where this plugin since version `1.1.0` is connecting to your battery inverter instead of your PV inverter?

Scream4Life, 2023-03-27:
It does somehow not recognize my inverter as a SMA inverter now. I'm on 1.1.1 with the hardcoded IP you've mentioned above. 

```
Mar 27 11:50:02 raspberrypi homebridge[21523]: [27.3.2023, 11:50:00] [homebridge-sma-home-manager] Discovered SMA inverter: no
Mar 27 11:50:02 raspberrypi homebridge[21523]: [27.3.2023, 11:50:00] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006124640, FirmwareRevision: '2.11.5.R' }
Mar 27 11:50:02 raspberrypi homebridge[21523]: [27.3.2023, 11:50:01] [homebridge-sma-home-manager] Discovered SMA inverter: { SerialNumber: 1980452352, FirmwareRevision: '0.19.0.N' }
Mar 27 11:50:02 raspberrypi homebridge[21523]: [27.3.2023, 11:50:01] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006124640, FirmwareRevision: '2.11.5.R' }
Mar 27 11:50:02 raspberrypi homebridge[21523]: [27.3.2023, 11:50:01] [homebridge-sma-home-manager] The discovered inverter is not a solar inverter! Please create a bug report with as much detail as possible.
Mar 27 11:50:19 raspberrypi homebridge[21523]: [27.3.2023, 11:50:19] [homebridge-sma-home-manager] This plugin is taking long time to load and preventing Homebridge from starting. See https://homebridge.io/w/JtMGR for more info.
Mar 27 11:50:39 raspberrypi homebridge[21523]: [27.3.2023, 11:50:39] [homebridge-sma-home-manager] This plugin is taking long time to load and preventing Homebridge from starting. See https://homebridge.io/w/JtMGR for more info.
Mar 27 11:51:59 raspberrypi homebridge[21523]: Error: Not running
Mar 27 11:51:59 raspberrypi homebridge[21523]:     at new NodeError (node:internal/errors:399:5)
Mar 27 11:51:59 raspberrypi homebridge[21523]:     at healthCheck (node:dgram:911:11)
Mar 27 11:51:59 raspberrypi homebridge[21523]:     at Socket.dropMembership (node:dgram:859:3)
Mar 27 11:51:59 raspberrypi homebridge[21523]:     at SMAHomeManager._keepMembershipActive (/usr/lib/node_modules/homebridge-sma-home-manager/index.js:803:15)
Mar 27 11:51:59 raspberrypi homebridge[21523]:     at listOnTimeout (node:internal/timers:569:17)
Mar 27 11:51:59 raspberrypi homebridge[21523]:     at processTimers (node:internal/timers:512:7)
Mar 27 11:51:59 raspberrypi homebridge[21523]: [27.3.2023, 11:51:59] [homebridge-sma-home-manager] Child bridge process ended
Mar 27 11:51:59 raspberrypi homebridge[21523]: [27.3.2023, 11:51:59] [homebridge-sma-home-manager] Process Ended. Code: 1, Signal: null
Mar 27 11:52:06 raspberrypi homebridge[21523]: [27.3.2023, 11:52:06] [homebridge-sma-home-manager] Restarting Process...
Mar 27 11:52:11 raspberrypi homebridge[21523]: [27.3.2023, 11:52:11] [homebridge-sma-home-manager] Launched child bridge with PID 22265
Mar 27 11:52:12 raspberrypi homebridge[21523]: [3/27/2023, 11:52:12 AM] Registering platform 'homebridge-sma-home-manager.SMAHomeManager'
Mar 27 11:52:12 raspberrypi homebridge[21523]: [27.3.2023, 11:52:12] [homebridge-sma-home-manager] Loaded homebridge-sma-home-manager v1.1.1 child bridge successfully
Mar 27 11:52:12 raspberrypi homebridge[21523]: [3/27/2023, 11:52:12 AM] Loaded 0 cached accessories from cachedAccessories.0EEC86B64944.
Mar 27 11:52:14 raspberrypi homebridge[21523]: [3/27/2023, 11:52:14 AM] [homebridge-sma-home-manager] Discovered SMA inverter: no
Mar 27 11:52:14 raspberrypi homebridge[21523]: [3/27/2023, 11:52:14 AM] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006124640, FirmwareRevision: '2.11.5.R' }
Mar 27 11:52:15 raspberrypi homebridge[21523]: [3/27/2023, 11:52:15 AM] [homebridge-sma-home-manager] Discovered SMA inverter: { SerialNumber: 1980452481, FirmwareRevision: '0.19.0.R' }
Mar 27 11:52:15 raspberrypi homebridge[21523]: [3/27/2023, 11:52:15 AM] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006124640, FirmwareRevision: '2.11.5.R' }
Mar 27 11:52:15 raspberrypi homebridge[21523]: [3/27/2023, 11:52:15 AM] [homebridge-sma-home-manager] The discovered inverter is not a solar inverter! Please create a bug report with as much detail as possible.
```
The devices are offline in EVE.

sweh, 2023-03-27:
> @sweh Do you also have the problem that @Scream4Life is seeing, where this plugin since version `1.1.0` is connecting to your battery inverter instead of your PV inverter?

No, it's connecting to the PV inverter as expected.

wimleers, 2023-03-27:
@sweh Thanks for confirming — that confirms my suspicion that @Scream4Life has a pretty atypical setup 😅

Scream4Life, 2023-03-27:
My pretty atypical setup just received a firmware update for my PV inverter to v3.12.1.R (latest). I also activated something like Webconnect I switched everything to auto-detect but the plugin still gathers my battery inverter instead. I still don't see the inverter with Discovery unter _http._tcp.

wimleers, 2023-03-30:
@sweh & @Scream4Life Can you both please update to https://github.com/wimleers/homebridge-sma-home-manager/releases/tag/1.1.3 and share the logged `DeviceType`? Thanks 🙏

sbf-, 2023-03-30:
> DeviceType: 9348

There's a huge list of types in the SBFspot taglists

`9348=STP10.0-3AV-40\0\Sunny Tripower 10.0`

Scream4Life, 2023-03-30:
> > @sweh & @Scream4Life Can you both please update to https://github.com/wimleers/homebridge-sma-home-manager/releases/tag/1.1.3 and share the logged `DeviceType`? Thanks 🙏
> 
> DeviceType: 9348

Without any code changes:
```
Mar 30 14:56:28 raspberrypi homebridge[13584]: [30.3.2023, 14:56:26] [homebridge-sma-home-manager] The discovered inverter is not a solar or hybrid inverter! Please create a bug report with as much detail as possible. {
Mar 30 14:56:28 raspberrypi homebridge[13584]:   energyManager: {
Mar 30 14:56:28 raspberrypi homebridge[13584]:     Model: 372,
Mar 30 14:56:28 raspberrypi homebridge[13584]:     SerialNumber: 3006124640,
Mar 30 14:56:28 raspberrypi homebridge[13584]:     FirmwareRevision: '2.11.5.R'
Mar 30 14:56:28 raspberrypi homebridge[13584]:   },
Mar 30 14:56:28 raspberrypi homebridge[13584]:   inverter: {
Mar 30 14:56:28 raspberrypi homebridge[13584]:     DeviceClass: 'battery inverter',
Mar 30 14:56:28 raspberrypi homebridge[13584]:     DeviceType: 9356,
Mar 30 14:56:28 raspberrypi homebridge[13584]:     SerialNumber: 3011693721,
Mar 30 14:56:28 raspberrypi homebridge[13584]:     FirmwareRevision: 'unknown'
Mar 30 14:56:28 raspberrypi homebridge[13584]:   }
Mar 30 14:56:28 raspberrypi homebridge[13584]: }
Mar 30 14:56:28 raspberrypi homebridge[13584]: [30.3.2023, 14:56:25] [homebridge-sma-home-manager] Discovered SMA inverter: no
Mar 30 14:56:28 raspberrypi homebridge[13584]: [30.3.2023, 14:56:25] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006124640, FirmwareRevision: '2.11.5.R' }
Mar 30 14:56:28 raspberrypi homebridge[13584]: [30.3.2023, 14:56:26] [homebridge-sma-home-manager] Discovered SMA inverter: {
Mar 30 14:56:28 raspberrypi homebridge[13584]:   DeviceClass: 'battery inverter',
Mar 30 14:56:28 raspberrypi homebridge[13584]:   DeviceType: 9356,
Mar 30 14:56:28 raspberrypi homebridge[13584]:   SerialNumber: 3011693721,
Mar 30 14:56:28 raspberrypi homebridge[13584]:   FirmwareRevision: 'unknown'
Mar 30 14:56:28 raspberrypi homebridge[13584]: }
Mar 30 14:56:28 raspberrypi homebridge[13584]: [30.3.2023, 14:56:26] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006124640, FirmwareRevision: '2.11.5.R' }
```

And with hardcoded IP in the _findInverter function:
```
Mar 30 15:03:10 raspberrypi homebridge[14729]: [30.3.2023, 15:03:09] [homebridge-sma-home-manager] Discovered SMA inverter: no
Mar 30 15:03:10 raspberrypi homebridge[14729]: [30.3.2023, 15:03:09] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006124640, FirmwareRevision: '2.11.5.R' }
Mar 30 15:03:11 raspberrypi homebridge[14729]: [30.3.2023, 15:03:10] [homebridge-sma-home-manager] Discovered SMA inverter: {
Mar 30 15:03:11 raspberrypi homebridge[14729]:   DeviceClass: 'solar inverter',
Mar 30 15:03:11 raspberrypi homebridge[14729]:   DeviceType: 9336,
Mar 30 15:03:11 raspberrypi homebridge[14729]:   SerialNumber: 1980452481,
Mar 30 15:03:11 raspberrypi homebridge[14729]:   FirmwareRevision: '0.19.0.N'
Mar 30 15:03:11 raspberrypi homebridge[14729]: }
Mar 30 15:03:11 raspberrypi homebridge[14729]: [30.3.2023, 15:03:10] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006124640, FirmwareRevision: '2.11.5.R' }
Mar 30 15:03:11 raspberrypi homebridge[14729]: [30.3.2023, 15:03:11] [homebridge-sma-home-manager] Inverter took 65.88641300052404 ms to respond, resulting in an invalid measurement. Dropping measurement.
Mar 30 15:03:11 raspberrypi homebridge[14729]: [30.3.2023, 15:03:11] [homebridge-sma-home-manager] Inverter took 210.54552400112152 ms to respond, resulting in an invalid measurement. Dropping measurement.
Mar 30 15:03:11 raspberrypi homebridge[14729]: [30.3.2023, 15:03:11] [homebridge-sma-home-manager] Inverter took 355.91999800130725 ms to respond, resulting in an invalid measurement. Dropping measurement.
Mar 30 15:03:11 raspberrypi homebridge[14729]: [30.3.2023, 15:03:11] [homebridge-sma-home-manager] Inverter took 477.3363820016384 ms to respond, resulting in an invalid measurement. Dropping measurement.
Mar 30 15:03:16 raspberrypi homebridge[14729]: [30.3.2023, 15:03:11] [homebridge-sma-home-manager] Inverter took 580.3370639979839 ms to respond, resulting in an invalid measurement. Dropping measurement.
Mar 30 15:03:16 raspberrypi homebridge[14729]: [30.3.2023, 15:03:13] [homebridge-sma-home-manager] Inverter took 1587.387197997421 ms to respond, resulting in an invalid measurement. Dropping measurement.
Mar 30 15:03:16 raspberrypi homebridge[14729]: [30.3.2023, 15:03:13] [homebridge-sma-home-manager] Inverter took 1411.4471959993243 ms to respond, resulting in an invalid measurement. Dropping measurement.
Mar 30 15:03:16 raspberrypi homebridge[14729]: [30.3.2023, 15:03:13] [homebridge-sma-home-manager] Inverter took 1404.1524679996073 ms to respond, resulting in an invalid measurement. Dropping measurement.
Mar 30 15:03:16 raspberrypi homebridge[14729]: [30.3.2023, 15:03:13] [homebridge-sma-home-manager] Inverter took 1270.2557369992137 ms to respond, resulting in an invalid measurement. Dropping measurement.
Mar 30 15:03:16 raspberrypi homebridge[14729]: [30.3.2023, 15:03:13] [homebridge-sma-home-manager] Inverter took 1292.443516999483 ms to respond, resulting in an invalid measurement. Dropping measurement.
Mar 30 15:03:16 raspberrypi homebridge[14729]: [30.3.2023, 15:03:13] [homebridge-sma-home-manager] Inverter took 1105.0471789985895 ms to respond, resulting in an invalid measurement. Dropping measurement.
Mar 30 15:03:16 raspberrypi homebridge[14729]: [30.3.2023, 15:03:15] [homebridge-sma-home-manager] Inverter took 1546.222362998873 ms to respond, resulting in an invalid measurement. Dropping measurement.
```

This way it is finally was showing production and consumption values again. But now it seems to freeze where as the log is this showing values with this "Dropping measurement" message.

wimleers, 2023-03-30:
Thanks, @sweh & @sbf-!

So, @Scream4Life sees device type 9356 for the battery inverter, 9336 for the solar inverter. According to @sbf-'s https://github.com/SBFspot/SBFspot/blob/master/SBFspot/TagListEN-US.txt that is
```
9356=SBS3.7-10\0\Sunny Boy Storage 3.7
```
and
```
9336=STP 15000TL-30\0\STP 15000TL-30
```

Now at least I can try to make sense of the ModBus registers documentation and try to find alternative addresses to read 👍  I now have users with similar problems:
- here in #5, @sweh with `9348` — but that seems a _hybrid_ inverter too, not a _battery_ inverter?
- here in #5, @Scream4Life with `9356` for a battery inverter
- over in #13, @andrejditz with a Sunny Tripower too … potentially the same one — awaiting his response at https://github.com/wimleers/homebridge-sma-home-manager/issues/13#issuecomment-1490315866. If it's the same, I'll merge that issue with this one.