wimleers/homebridge-sma-home-manager #6: Inverted Surplus Calculation

bodomenke, 2023-03-11:
@wimleers First of all: Great work. Thanks!

It seems there is an issue with the calculation of surpluses or at least in the label of the surplus accessories: it is shown as <= instead of >=…

Pls see screenshot. 
[image]

wimleers, 2023-03-11:
Oh wow, this was literally your first GitHub issue?! Welcome! 😄
[image]

bodomenke, 2023-03-13:
> Glad you like it!
> 
> Actually that `<=` is intentional: that's showing you the _reason_ why it's currently _off_ ("Nein" in German apparently!).
> 
> As soon as the energy signal turns on, it'll change 😊
> 
> The [README](https://github.com/wimleers/homebridge-sma-home-manager/blob/main/README.md) says:
> 
> > All energy signals have an On/Off status with a Reason to make it easy to understand why this energy signal is on or off.
> 
> Can you suggest how I should phrase that differently so that it's clear for the next person? 🙏

I guess it is a matter of how one reads the Eve app screen and how one decodes logical expressions. 😉

I do it like this:

Surplus <= X: Nein

-> results for me in logical consequence:
Surplus > X: Ja

But I got you. The „Nein“ relates to the name of the switch, which should ideally read like „500+ Surplus“ or so and the subline is simply the explanation, as you already stated. 

So, I guess it is my reading failure. I will rename my switches and all is fine. 😊

bodomenke, 2023-03-13:
> Oh wow, this was literally your first GitHub issue?! Welcome! 😄

Yeah 😊😎
Thank you 👍

wimleers, 2023-03-13:
Alright! 😊

Feel free to let me know a rephrasing suggestion when you think of a better way to explain it!

closed 2023-03-13 by wimleers