wimleers/homebridge-sma-home-manager #7: Home Manager 1.0 or Inverter only compatibility

solbadguy2010, 2023-03-13:
**Is your feature request related to a problem? Please describe:**

**Describe the solution you'd like:**
I would like to "disable" the need for a Home Manager 2.0. For me it would be enough to only use direct modbus communication with the inverter to see my live data in homekit.

**Describe alternatives you've considered:**
I used the https://github.com/codyc1515/homebridge-sma-inverter plugin until now, but i would love to use an modern and updatetd one

**Additional context:**
**Inverter:**
* **Model:** SMA Sunny Tripower 8000 TL-20

solbadguy2010, 2023-03-13:
I currently receive this error when i try to start your Plugin:

[13/03/2023, 10:31:57] [homebridge-sma-home-manager] Launched child bridge with PID 16943
[13/03/2023, 10:31:57] Registering platform 'homebridge-sma-home-manager.SMAHomeManager'
[13/03/2023, 10:31:57] [homebridge-sma-home-manager] Loaded homebridge-sma-home-manager v1.0.1 child bridge successfully
[13/03/2023, 10:31:58] [homebridge-sma-home-manager] Restarting child bridge...
[13/03/2023, 10:31:58] Got SIGTERM, shutting down child bridge process...

/usr/local/lib/node_modules/homebridge-sma-home-manager/index.js:330
                                serialNumber = data.buffer.readUInt32BE();
                        ^
TypeError: Cannot read properties of undefined (reading 'buffer')
    at SMAHomeManager.<anonymous> (/usr/local/lib/node_modules/homebridge-sma-home-manager/index.js:330:25)
    at ModbusRTU.writeFC4 (/usr/local/lib/node_modules/homebridge-sma-home-manager/node_modules/modbus-serial/index.js:544:19)
    at ModbusRTU.writeFC3 (/usr/local/lib/node_modules/homebridge-sma-home-manager/node_modules/modbus-serial/index.js:530:10)
    at ModbusRTU.converted [as readHoldingRegisters] (/usr/local/lib/node_modules/homebridge-sma-home-manager/node_modules/modbus-serial/apis/promise.js:36:27)
    at SMAHomeManager._readInverterMetadata (/usr/local/lib/node_modules/homebridge-sma-home-manager/index.js:329:11)
    at SMAHomeManager.<anonymous> (/usr/local/lib/node_modules/homebridge-sma-home-manager/index.js:206:8)
    at listOnTimeout (node:internal/timers:564:17)
    at processTimers (node:internal/timers:507:7)
[13/03/2023, 10:31:58] [homebridge-sma-home-manager] Child bridge process ended
[13/03/2023, 10:31:58] [homebridge-sma-home-manager] Process Ended. Code: 1, Signal: null
[13/03/2023, 10:32:04] [homebridge-sma-home-manager] Stopping child bridge (will not restart)...
[13/03/2023, 10:32:14] [homebridge-sma-home-manager] Bridge already shutting down or stopped.

wimleers, 2023-03-14:
- I will not add support for an “inverter only” mode. See https://github.com/wimleers/homebridge-sma-home-manager/issues/2#issuecomment-1452285495 — plus I don’t want to maintain even more than what this already does!
- Do you actually have a Home Manager 1.0?

Why not just continue to use that old plugin? It’s probably good enough?

solbadguy2010, 2023-03-14:
I want to have a plugin that is in active development.
Yes, i have a Home Manager 1.0. Maybe its possible to get this one compatible with your plugin?

wimleers, 2023-03-14:
Yes, quite possibly!

If you could collect all relevant technical documentation, that would definitely speed up the process of getting an answer to that! 😊

solbadguy2010, 2023-03-14:
What do you need? Please also look in Twitter Chat :-)

wimleers, 2023-03-15:
What's Twitter Chat? 😅

I need the same kind of technical documentation for the SMA Home Manager 1.0 as exists for SMA Home Manager 2.0 and SMA Energy Meter: https://developer.sma.de/fileadmin/content/global/Partner/Documents/SMA_Labs/EMETER-Protokoll-TI-en-10.pdf — which is linked from the `README.md` of this project.

wimleers, 2023-03-15:
Note that adding support for an inverter-only mode is not only a pain because of https://github.com/wimleers/homebridge-sma-home-manager/issues/7#issuecomment-1468301107, but also because the current plugin logic actually relies _less_ on reading data from the inverter than https://github.com/codyc1515/homebridge-sma-inverter did, because much of the necessary data arrives every second already anyway from the Home Manager's Speedwire multicasting!

In other words: this would add significant complexity. And it'd increase the communication rate with the inverter via Modbus, which has a limited communication frequency _and_ is rather sensitive/error-prone to concurrent requests.

For that reason, I just retitled this to `Home Manager 1.0 support`.

solbadguy2010, 2023-03-15:
Twitter PM? I wrote you a messagt there :-)

Home Manager 1.0 has no build in Power Meter functionality.

solbadguy2010, 2023-03-15:
By the way, do you know this URL?
https://www.sunnyportal.com/homemanager
You need to login to your Sunny Portal first. Maybe a way to create an online version of the plugin...

wimleers, 2023-03-15:
> Twitter PM? I wrote you a messagt there :-)

Ah! I never check that 😅

> Home Manager 1.0 has no build in Power Meter functionality.

Ah 😬 I just was able to confirm that on the official docs at https://my.sma-service.com/s/article/Controlling-Active-Power-Limitation-With-Sunny-Home-Manager-1-0?language=en_US 👍

That means it cannot possibly work … but perhaps you have an [SMA Energy Meter](https://www.sma.de/en/products/monitoring-control/sma-energy-meter)?

> By the way, do you know this URL? https://www.sunnyportal.com/homemanager You need to login to your Sunny Portal first. Maybe a way to create an online version of the plugin...

I did not! That currently returns this for me:
```
{
  "__type": "LiveDataUI",
  "Timestamp": {
    "__type": "DateTime",
    "DateTime": "2023-03-15T15:26:26",
    "Kind": "Unspecified"
  },
  "TimestampUtc": {
    "__type": "DateTime",
    "DateTime": "2023-03-15T14:26:26",
    "Kind": "Utc"
  },
  "PV": 1372,
  "FeedIn": 698,
  "GridConsumption": 0,
  "DirectConsumption": null,
  "SelfConsumption": 674,
  "SelfSupply": 674,
  "TotalConsumption": 674,
  "DirectConsumptionQuote": null,
  "SelfConsumptionQuote": 49,
  "AutarkyQuote": 100,
  "BatteryIn": null,
  "BatteryOut": null,
  "BatteryChargeStatus": null,
  "OperationHealth": null,
  "BatteryStateOfHealth": null,
  "ModuleTemperature": null,
  "EnvironmentTemperature": null,
  "WindSpeed": null,
  "Insolation": null,
  "BatteryMode": null,
  "InfoMessages": [
    
  ],
  "WarningMessages": [
    
  ],
  "ErrorMessages": [
    
  ],
  "Info": {
    
  }
}
```
… which is both inaccurate and incomplete 🙈

On top of that, this would introduce:
- a cloud dependency … (very bad! I want this plugin to work even without internet access, just like every officially HomeKit-certified device)
- … on a non-official API (very bad! that means it can stop working at any point)

solbadguy2010, 2023-03-15:
I have no SMA Energy Meter but a third party one. My sunny portal knows how much power the house uses in total.

sorry, that the api is not helping. My idea was maybe create a second (online only) plugin :-)

wimleers, 2023-03-16:
How does that third party one communicate with Sunny Portal?

solbadguy2010, 2023-03-16:
Via Bus wires i think
[image]
[image]

wimleers, 2023-03-16:
Woah! Yeah. That looks ancient compared to the Home Manager 2.0 😅 How old is this setup?

Note that this plugin _should_ work just fine with the [SMA Energy Meter](https://www.sma.de/en/products/monitoring-control/sma-energy-meter) too. The cost of that is ~1/3rd lower than that of an SMA Home Manager 2.0. If you really want to use this plugin, I suggest you update to that.

I do not have access to a Home Manager 1.0, so even if I had the time, I would never be able to add support for this.

closed 2023-03-16 by wimleers

solbadguy2010, 2023-03-16:
It was build 2013 :-D

My dad has an Home Manager 1.0 + SMA Energy Meter setup. I can test this and give you a feedback.

Please have a last look on Twitter PM ;-)

wimleers, 2023-03-16:
That sounds awesome! Looking forward to hearing from you, please create a new issue!