wimleers/homebridge-sma-home-manager #8: [Support] Log Overview

Scream4Life, 2023-03-16:
Hi, as recommended I installed the plugin as child bridge. Nevertheless the log of the plugin is pushing its "news" in the overall log of the parent bridge. Is there any chance to separate the information? Because identifying errors of other plugins is quite hard now.

Scream4Life, 2023-03-16:
...and with logging I mean the output that comes from this call which I basically use for requesting what's going on

sudo journalctl -fau homebridge

wimleers, 2023-03-16:
Homebridge centralizes the logs of all child bridges into a single overall log. I've always found this to be exactly what I expected. That's not the case for you?

Also: are you seeing so many errors then that this is a problem? 🤯 From my plugin or from others?

wimleers, 2023-03-16:
BTW, when you update to https://github.com/wimleers/homebridge-sma-home-manager/releases/tag/1.1.0, you won't have to modify the code anymore to hardcode your IP address! 🥳

Scream4Life, 2023-03-16:
Looking forward to this release. Can't see it in config UI yet.

Nice to know that I'm running homebridge in debug mode 😉 That explains something. How can I deactivate that? I did the setup from a manual with several instances but I'm unable to find my mistake.

I'm starting the homebridge with this **sudo systemctl restart homebridge -l** 

/etc/systemd/system/homebridge.service
```
[Unit]
Description=homebridge server
After=syslog.target network-online.target

[Service]
Type=simple
User=pi
EnvironmentFile=/etc/default/homebridge
ExecStart=/usr/bin/homebridge -U $HOMEBRIDGE_OPTS
Restart=on-failure
RestartSec=10 KillMode=process
SuccessExitStatus=143

[Install]
WantedBy=multi-user.target
```

/etc/default/homebridge
```
# Defaults / Configuration options for homebridge
# The following settings tells homebridge where to find the config.json file and where to persist the data (i.e. pairing and others)
HOMEBRIDGE_OPTS=-I -U /var/homebridge
# If you uncomment the following line, homebridge will log more
# You can display this via systemd's journalctl: journalctl -f -u homebridge # DEBUG=*

# To enable web terminals via config-ui-x uncomment the following line
HOMEBRIDGE_CONFIG_UI_TERMINAL=1
```

wimleers, 2023-03-16:
[image]

Scream4Life, 2023-03-16:
I wish it were that simple

[image]

wimleers, 2023-03-16:
What version of Homebridge is that?

Scream4Life, 2023-03-16:
Homebridge is 1.6.0 
Homebridge UI 4.50.2

wimleers, 2023-03-19:
Same versions here.

Questions:
1. How did you install Homebridge? I’m shocked to see the UI is so different 🤯
2. Are you talking about the log as displayed by Homebridge’s UI, or the `syslog` of Linux?

Scream4Life, 2023-03-19:
1) Is was back in 2017, before the Config UI-X time 😃 Installed everything the hard way. 
2) It is the log when I look on "sudo journalctl -fau homebridge" when I restart the homebridge service.

wimleers, 2023-03-19:
1. Okay!
2. Ah. So it’s basically your OS’ `syslog`. AFAIK all levels are sent to that log. That’s a Homebridge concept/responsibility, not one of this plugin.

It’s because you titled this “Log Overview” that I was led to think that this was about Homebridge’s overview UI. That wasn’t the case then?

wimleers, 2023-03-19:
See `journalctl -p <number or text priority>`.

Scream4Life, 2023-03-19:
Ah, I see. Thank you. I‘m looking out for a real homebridge log now, not that syslog.

closed 2023-03-19 by Scream4Life