wimleers/homebridge-sma-home-manager #9: [Bug] characteristic was supplied illegal value

Scream4Life, 2023-03-16:
I'm running your new 1.0.2 release now, changed the inverter IP again to my local IP in the index.js but I'm getting these error messages?

**Paste of Logs:**
```
Mar 16 09:14:38 raspberrypi homebridge[24765]: [Live power flow@Power Monitor@Consumption] characteristic was supplied illegal value: number -846.9000000000001 exceeded minimum of 0
Mar 16 09:14:38 raspberrypi homebridge[24765]: [Live power flow@Power Monitor@Production] characteristic was supplied illegal value: number -870 exceeded minimum of 0
Mar 16 09:14:38 raspberrypi homebridge[24765]: [Recent power flow@Power Monitor@Consumption] characteristic was supplied illegal value: number -881.4000000000001 exceeded minimum of 0
Mar 16 09:14:38 raspberrypi homebridge[24765]: [Recent power flow@Power Monitor@Production] characteristic was supplied illegal value: number -882.3000000000001 exceeded minimum of 0
Mar 16 09:14:38 raspberrypi homebridge[24765]: [16.3.2023, 09:14:38] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Consumption': characteristic was supplied illegal value: number -846.9000000000001 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 16 09:14:38 raspberrypi homebridge[24765]: [16.3.2023, 09:14:38] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Production': characteristic was supplied illegal value: number -870 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 16 09:14:38 raspberrypi homebridge[24765]: [16.3.2023, 09:14:38] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Consumption': characteristic was supplied illegal value: number -881.4000000000001 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 16 09:14:38 raspberrypi homebridge[24765]: [16.3.2023, 09:14:38] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Production': characteristic was supplied illegal value: number -882.3000000000001 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
```

Any idea what's causing this?

Minutes ago the devices were showing no production and consumption. Now the value are there again. Weird.

wimleers, 2023-03-16:
I see you have Homebridge's Debug mode on! That'd explain why you see so many messages as you reported in #8…

Still these warnings should not occur obviously. I have not seem them before, but will turn on Debug mode again to see if I can reproduce them too.

Thanks for the report!

wimleers, 2023-03-16:
Also: note that this is harmless — it just means that these particular updates _failed_ to update the characteristics, i.e. you didn't get the update for that single second. If this happens even just a few dozen times per day, it still means the other >86000 seconds per day succeeded 🤓

Still, I'd like this to be fixed too of course!

wimleers, 2023-03-17:
# ⚠️ Do not trust `sunnyportal.com`!

http://sunnyportal.com/ is _terrible_ at displaying live information.

It's by definition not "live", since it first has to go from your local network to their servers, to be processed, and then sent back to you, so a ~0.5 second lag would be _great_, but a multi-second lag is closer to reality.

Just compare it with your inverter's local web UI. You'll see things like this:
[image]

Same thing for the SMA Energy app by the way. Just turn on a big consumer, like a hair dryer. Observe how long it takes to see that ~2000 W spike in the Eve  app (using my Homebridge plugin) versus `sunnyportal.com` versus the SMA Energy app. I'm confident it'll be visible first in Eve!

# Battery

I suspect that what you're seeing is the behavior of your battery inverter trying to maintain a 0W import/0W export level. Your SMA Sunny Island battery inverter is constantly trying to match the needs. I'm not sure how often, but probably _at least_ once per second, potentially more often. Still, it's impossible to get perfectly right, and hence I'd expect to see this kind of oscillation.

You've just never been able to even _see_ this before because neither `sunnyportal.com` nor the SMA energy app display information in real time. 😅

I suspect that on setups with a battery inverter, the _Live power flow_ accessory I'm exposing is too noisy to be usable. I suspect that we need to average it out over at least 2 seconds, possibly a few more.

**Can you please check the _Recent power flow_ accessory? 🤞 That averages things out over the last 3 minutes, and probably shows 0W import in your case!**

The problem is I don't have a battery inverter, so it's hard for me to even implement #5. Unless you're willing to give me remote access … so I can develop on _your_ machine and observe the behavior. 🙃

Scream4Life, 2023-03-17:
That‘s interesting. Thanks for that eye opener. 

Today every time I checked the production value was zero, while the sun was shining. I‘m wondering why this sometimes works and sometimes it doesn‘t. Maybe something broke with the last release? Checking tomorrow again. 
At the moment the production equals the consumption, because the battery usage is seen as production. Nice to know. 

Even the Totals look weird today. 2,1kwh for the whole day while the App says 26,4kwh and the battery was full. At least Import and Export looked close to the App value. 

What do you need for a remote session to check the battery inverter output? And how long do you think does it take? I‘m wondering whether this is doable.

Scream4Life, 2023-03-17:
lol And no-sun is one ‚No’ right now because the battery is delivering a production value, I guess.

wimleers, 2023-03-18:
Could you please browse to https://169.254.12.3/ while on your local network and let me know if it’s the Solar inverter or the battery that is responding?

Scream4Life, 2023-03-18:
It doesn't load at all. I was actually never able to load the PV inverter IP and see a web interface. But I can load the Battery Inverter interface with an IP that I actually thought was the PV inverter IP. Hmpf

Scream4Life, 2023-03-18:
Some debug line from the homebridge?! Production stays zero. Weird that it looked good in the evening. Is it pulling the values from the battery inverter? Any chance to manually input the correct IP again to see whether that's the problem?

```
Mar 18 10:03:00 raspberrypi homebridge[3426]: [Recent power flow@Power Monitor@Production] characteristic was supplied illegal value: number -2551.4 exceeded minimum of 0
Mar 18 10:03:00 raspberrypi homebridge[3426]: [Live power flow@Power Monitor@Consumption] characteristic was supplied illegal value: number -2531.9 exceeded minimum of 0
Mar 18 10:03:00 raspberrypi homebridge[3426]: [Live power flow@Power Monitor@Production] characteristic was supplied illegal value: number -2540 exceeded minimum of 0
Mar 18 10:03:00 raspberrypi homebridge[3426]: [Recent power flow@Power Monitor@Consumption] characteristic was supplied illegal value: number -2547.8 exceeded minimum of 0
Mar 18 10:03:00 raspberrypi homebridge[3426]: [Recent power flow@Power Monitor@Production] characteristic was supplied illegal value: number -2551.2000000000003 exceeded minimum of 0
Mar 18 10:03:00 raspberrypi homebridge[3426]: [18.3.2023, 10:02:59] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Consumption': characteristic was supplied illegal value: number -2551.6000000000004 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 10:03:00 raspberrypi homebridge[3426]: [18.3.2023, 10:02:59] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Production': characteristic was supplied illegal value: number -2547 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 10:03:00 raspberrypi homebridge[3426]: [18.3.2023, 10:02:59] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Consumption': characteristic was supplied illegal value: number -2548.7000000000003 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 10:03:00 raspberrypi homebridge[3426]: [18.3.2023, 10:02:59] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Production': characteristic was supplied illegal value: number -2551.2000000000003 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 10:03:01 raspberrypi homebridge[3426]: [18.3.2023, 10:02:59] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Consumption': characteristic was supplied illegal value: number -2552.8 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 10:03:01 raspberrypi homebridge[3426]: [18.3.2023, 10:02:59] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Production': characteristic was supplied illegal value: number -2547 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 10:03:01 raspberrypi homebridge[3426]: [18.3.2023, 10:02:59] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Consumption': characteristic was supplied illegal value: number -2548.8 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 10:03:01 raspberrypi homebridge[3426]: [18.3.2023, 10:02:59] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Production': characteristic was supplied illegal value: number -2551.1000000000004 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 10:03:01 raspberrypi homebridge[3426]: [18.3.2023, 10:02:59] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Consumption': characteristic was supplied illegal value: number -2578.2000000000003 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 10:03:01 raspberrypi homebridge[3426]: [18.3.2023, 10:02:59] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Production': characteristic was supplied illegal value: number -2550 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 10:03:01 raspberrypi homebridge[3426]: [18.3.2023, 10:02:59] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Consumption': characteristic was supplied illegal value: number -2549.4 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 10:03:01 raspberrypi homebridge[3426]: [18.3.2023, 10:02:59] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Production': characteristic was supplied illegal value: number -2551.1000000000004 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 10:03:01 raspberrypi homebridge[3426]: [18.3.2023, 10:02:59] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Consumption': characteristic was supplied illegal value: number -2552.4 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 10:03:01 raspberrypi homebridge[3426]: [18.3.2023, 10:02:59] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Production': characteristic was supplied illegal value: number -2550 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
[... cut ...]
Mar 18 10:03:01 raspberrypi homebridge[3426]: [18.3.2023, 10:02:59] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Consumption': characteristic was supplied illegal value: number -2549.6000000000004 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 10:03:01 raspberrypi homebridge[3426]: [18.3.2023, 10:02:59] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Production': characteristic was supplied illegal value: number -2550 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.

```

wimleers, 2023-03-18:
> Is it pulling the values from the battery inverter?

That is my suspicion.

> Any chance to manually input the correct IP again to see whether that's the problem?

Please update to version 1.1.0. In that version, I’m doing automatic discovery of existing devices. It should find the inverter based on the advertised service name.

Scream4Life, 2023-03-18:
I was on 1.1.0. That caused the problem. Now I‘m back on the previous version in which I inserted the correct IP in the index.js. Works correct with that.

Scream4Life, 2023-03-18:
We've been on this road already 😉  But here are the outputs...just the Battery Inverter and the Home Manager. My PV Inverter does not show up here at HTTP.

[image]

[image]

wimleers, 2023-03-18:
Wasn’t sure if that was you! 😅

I also need:
- screenshot of what you see after tapping the discovered inverter
- Homebridge log of when this plugin is booting until it has launched all its accessories

Those pieces of information will show me which IP address is the right one and which one it actually connected to.

Scream4Life, 2023-03-18:
[image]

```
Mar 18 09:50:27 raspberrypi homebridge[1297]: [18.3.2023, 09:50:27] ---
Mar 18 09:50:28 raspberrypi homebridge[1297]: [18.3.2023, 09:50:28] Loaded plugin: homebridge-sma-home-manager@1.1.0
Mar 18 09:50:28 raspberrypi homebridge[1297]: [18.3.2023, 09:50:28] Registering platform 'homebridge-sma-home-manager.SMAHomeManager'
Mar 18 09:50:28 raspberrypi homebridge[1297]: [18.3.2023, 09:50:28] ---
Mar 18 09:51:06 raspberrypi homebridge[1297]: [18.3.2023, 09:51:06] [SMAHomeManager] Initializing SMAHomeManager platform...
Mar 18 09:51:06 raspberrypi homebridge[1297]: [18.3.2023, 09:51:06] [SMAHomeManager] Initializing child bridge 0E:EC:86:B6:49:44
Mar 18 09:51:06 raspberrypi homebridge[1297]: [18.3.2023, 09:51:06] Loading 12 accessories...
Mar 18 09:51:28 raspberrypi homebridge[1297]: [18.3.2023, 09:51:28] [homebridge-sma-home-manager] Launched child bridge with PID 1524
Mar 18 09:51:29 raspberrypi homebridge[1297]: [18.3.2023, 09:51:29] [homebridge-sma-home-manager] Loaded homebridge-sma-home-manager v1.1.0 child bridge successfully
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:28] Registering platform 'homebridge-sma-home-manager.SMAHomeManager'
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:31] [homebridge-sma-home-manager] Discovered SMA inverter: no
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:31] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006124640, FirmwareRevision: '2.11.5.R' }
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:32] [homebridge-sma-home-manager] Discovered SMA inverter: { SerialNumber: 3011693721, FirmwareRevision: '3.18.12.R' }
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:32] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006124640, FirmwareRevision: '2.11.5.R' }
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:32] Initializing platform accessory 'Live power flow'...
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:32] Initializing platform accessory 'Recent power flow'...
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:32] Initializing platform accessory 'Today's energy flow'...
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:32] Initializing platform accessory 'Off Grid energy signal'...
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:32] Initializing platform accessory 'No Sun energy signal'...
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:34] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Consumption': characteristic was supplied illegal value: number -2778.6000000000004 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:34] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Production': characteristic was supplied illegal value: number -2743 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:34] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Consumption': characteristic was supplied illegal value: number -2778.6000000000004 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:34] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Production': characteristic was supplied illegal value: number -2743 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:35] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Consumption': characteristic was supplied illegal value: number -2749.6000000000004 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:35] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Production': characteristic was supplied illegal value: number -2743 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:35] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Consumption': characteristic was supplied illegal value: number -2764.1000000000004 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:35] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Production': characteristic was supplied illegal value: number -2743 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:35] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Consumption': characteristic was supplied illegal value: number -2723.6000000000004 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:35] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Production': characteristic was supplied illegal value: number -2743 exceeded minimu [... cut ...]
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:35] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Production': characteristic was supplied illegal value: number -2682 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
Mar 18 09:51:43 raspberrypi homebridge[1297]: [18.3.2023, 09:51:35] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Consumption': characteristic was supplied illegal value: number -2687.9 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.

```

Scream4Life, 2023-03-18:
1) Weird. As I said I'm currently on 1.0.2 with my local IP. The whole day has shown reasonable figures. For consumption, production, import and export. Now in the evening there is no production and also no consumption. At least the last one is strange. Maybe because the inverter does not get started by the sun-energy?!

2) Yesterday with 1.1.0 and probably the battery inverter data the production and the consumption was almost equal because the battery delivered 100% of the required energy - which was shown as consumption. 

Looks 1) normal for you? If yes another device in the energy room would be nice "battery power flow". Maybe a fix IP in the beginning to test this 😃 Obviously your plugin can read this data already.

wimleers, 2023-03-18:
That's really odd, you should have seen something like this in your log:
```
[3/18/2023, 9:22:59 PM] [homebridge-sma-home-manager] Found inverter SMA3010955555 at SMA3010955555.local with IPv4 address 192.168.0.237.
[3/18/2023, 9:22:59 PM] [homebridge-sma-home-manager] Discovered SMA inverter: no
[3/18/2023, 9:22:59 PM] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3012888888, FirmwareRevision: '2.11.5.R' }
[3/18/2023, 9:23:00 PM] [homebridge-sma-home-manager] Discovered SMA inverter: { SerialNumber: 3010955555, FirmwareRevision: '4.0.73.R' }
[3/18/2023, 9:23:00 PM] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3012888888, FirmwareRevision: '2.11.5.R' }
```

Questions:

1. Any chance you stripped that part away? Because it's that `Found inverter` message that contains the crucial information. It should be in _yellow_ because it's a _warning_ message.
2. What inverter model do you have? SMA's site seems to have the worst search engine support of any site I've ever seen because your firmware version is impossible to find using any browser 
3. What is the firmware version that is reported by SMA itself?

Note: adding battery support is out of scope for this issue, that's covered by #5. The scope of this issue is to make sure that the numbers look correct even for systems that have a battery inverter installed.

wimleers, 2023-03-18:
As I was trying to figure out the answer to question 2 above myself, I just noticed that the `4.0.73.R` that I'm seeing for my inverter is actually the "Firmware version of the main processor", not the "Firmware version of the logic component", which is `4.0.75.R`, [which is what was recently automatically updated by SMA](https://github.com/wimleers/homebridge-sma-home-manager/issues/4#issuecomment-1468669369). Turns out that for SMA's main consumer-facing inverter productline, it's _impossible_ to read the actual firmware 😬🤯

The list of inexplicable limitations is definitely growing…

wimleers, 2023-03-18:
> As I was trying to figure out the answer to question 2 above myself, I just noticed that the `4.0.73.R` that I'm seeing for my inverter is actually the "Firmware version of the main processor", not [...]

Rather than lying, as of https://github.com/wimleers/homebridge-sma-home-manager/commit/d689c7ca607dd6604e8f1c7400223c3ea00b64a6, the firmware version `unknown` will be concluded for inverters that do not support it. 👍

Scream4Life, 2023-03-19:
DIdn't see these lines.
```
Mar 18 10:01:36 raspberrypi homebridge[3426]: [18.3.2023, 10:01:36] ---
Mar 18 10:01:36 raspberrypi homebridge[3426]: [18.3.2023, 10:01:36] Loaded plugin: homebridge-sma-home-manager@1.1.0
Mar 18 10:01:36 raspberrypi homebridge[3426]: [18.3.2023, 10:01:36] Registering platform 'homebridge-sma-home-manager.SMAHomeManager'
Mar 18 10:01:36 raspberrypi homebridge[3426]: [18.3.2023, 10:01:36] ---
Mar 18 10:02:54 raspberrypi homebridge[3426]: [18.3.2023, 10:02:48] [homebridge-sma-home-manager] Discovered SMA inverter: no
Mar 18 10:02:54 raspberrypi homebridge[3426]: [18.3.2023, 10:02:48] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006124640, FirmwareRevision: '2.11.5.R' }
Mar 18 10:02:54 raspberrypi homebridge[3426]: [18.3.2023, 10:02:49] [homebridge-sma-home-manager] Discovered SMA inverter: { SerialNumber: 3011693721, FirmwareRevision: '3.18.12.R' }
Mar 18 10:02:54 raspberrypi homebridge[3426]: [18.3.2023, 10:02:49] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006124640, FirmwareRevision: '2.11.5.R' }
```

It seems my inverter is not auto-updating. I have the Sunny Tripower 15000TL-30
Firmware-Version of Communication group: 2.63.0.R  
Firmware-Version of Main processor: 1.02.2.R  
Firmware-Version of Logic component: 0.13.0.R  
Firmware-Version: 2.83.3.R

wimleers, 2023-03-26:
I can find your PV inverter at https://files.sma.de/downloads/STP15-25TL-30-DS-en-41.pdf but not in the download center: https://www.sma.de/en/service/downloads — ah because it's no longer sold, so I had to look under "archive"! Depending on circumstances, the latest version available to you is [2.90.01.R](https://files.sma.de/downloads/FW2-90-01-R_STP15_20_25000TL-30.zip) or [3.12.01.R](https://files.sma.de/downloads/FW3-12-01-R_STP15_20_25000TL-30.zip), both released May 23, 2022. So an update is for sure available.

wimleers, 2023-03-26:
It seems like you're monitoring the log using something else than Homebridge's own UI. That's fine, but then it depends on the filtering/configuration you have in place. It really should not be possible for you to not have that 
```
[3/18/2023, 9:22:59 PM] [homebridge-sma-home-manager] Found inverter SMA3010955555 at SMA3010955555.local with IPv4 address 192.168.0.237.
```
line. It's logged as a `warning`.

wimleers, 2023-03-26:
I've not been able to reproduce this _yet_, so I'd stil like to try to verify that you're indeed connected to the PV inverter and not the battery inverter. Since I don't have a battery inverter nor do solid docs seem to be available online, I do not know its behavior… 😞

Is this happening all the time? If not: when? (Surplus production, no production, battery discharging vs not, et cetera.)

wimleers, 2023-03-26:
Taking a step back away from the concrete situation, let's try to determine the root cause by reasoning from the opposite end: the code.

Note that you're getting the errors only for **live/recent production**  and **live/recent consumption**, not for the "today" numbers.

The relevant code:
```
			// Read the live production from the inverter, to minimize the time offset
			// relative to the received energy manager datagram (S32, FIX0: 0 decimals).
			const before = performance.now();
			client.readHoldingRegisters(30775, 2).then((data) => {
				return data.buffer.equals(SMA_MODBUS_S32_NAN_VALUE) ? 0 : data.buffer.readInt32BE();
			})
			.then((producedWatts) => {
				const estimatedMsOffset = performance.now() - before;
				const [timestamp, netWatts] = this._parseDatagram(msg, rinfo);
				this._processMeasurement(producedWatts, netWatts, timestamp);
			})
```
and
```
 	 _ p r o c e s s M e a s u r e m e n t :   f u n c t i o n ( p r o d u c e d W a t t s F r o m I n v e r t e r ,   n e t W a t t s F r o m E n e r g y M a n a g e r ,   t i m e s t a m p F r o m E n e r g y M a n a g e r )   { 
 	 	 / /   C a p t u r e   t h e   o b s e r v a t i o n s   i n   a   " m e a s u r e m e n t "   o b j e c t . 
 	 	 c o n s t   i m p o r t W a t t s   =   n e t W a t t s F r o m E n e r g y M a n a g e r   >   0   ?   n e t W a t t s F r o m E n e r g y M a n a g e r :   0 ; 
 	 	 c o n s t   e x p o r t W a t t s   =   n e t W a t t s F r o m E n e r g y M a n a g e r   <   0   ?   - n e t W a t t s F r o m E n e r g y M a n a g e r :   0 ; 
 	 	 c o n s t   m e a s u r e m e n t   =   { 
 	 	 	 t i m e s t a m p :   t i m e s t a m p F r o m E n e r g y M a n a g e r , 
 	 	 	 i m p o r t :   i m p o r t W a t t s , 
 	 	 	 e x p o r t :   e x p o r t W a t t s , 
 	 	 	 p r o d u c t i o n :   p r o d u c e d W a t t s F r o m I n v e r t e r , 
 	 	 	 c o n s u m p t i o n :   i m p o r t W a t t s   +   p r o d u c e d W a t t s F r o m I n v e r t e r   -   e x p o r t W a t t s , 
 	 	 } ;
```

So `producedWatts`/`producedWattsFromInverter` is literally was read from the inverter, no processing has been applied. That number should NEVER be negative, since no power can flow _into_ the solar panels.

But … the ModBus register 30775 is `S32`, meaning "signed 32-bit integer". So it's allowed to become negative in theory. Which definitely could make sense for a _battery inverter_!

AFAICT this makes my theory that you're actually connected to the battery inverter instead of the solar inverter more likely! 🤓 So now I _really_ need your equivalent of
```
[3/18/2023, 9:22:59 PM] [homebridge-sma-home-manager] Found inverter SMA3010955555 at SMA3010955555.local with IPv4 address 192.168.0.237.
```
to know this for sure.

wimleers, 2023-03-26:
Well actually, looking back:
- you shared your DNS-SD discovery results and said that `SMA3011693721.local` is your battery inverter, not your PV inverter
- the log literally says it's connected to `3011693721`

So we've got our conclusion. So this really just is a duplicate of #5. Let's continue there.

Scream4Life, 2023-03-26:
Well, #5 I requested an option to read battery inverter details. 

This one here was created because version 1.1.0 obviously picked my battery inverter automatically and tried to show some data of that. Maybe I should run 1.1.0 and 1.0.2 on different instances to have the best of both worlds 😉 

Currently I'm running 1.0.2 with the manual input of my PV inverters IP address (I wasn't able to insert his in 1.1.0 manually). With this version I can see on daytimes the consumption of the house + battery income (if battery level is < <100%) as consumption value and the production-value is correct income from the sun. At night both values are zero probably because the PV inverter does not get energy from the sun to calculate any value. It's off! It seems that I need 1.1.0 on nighttimes since it was showing reasonable data there 😉

wimleers, 2023-03-26:
Actually … we still need to figure out here then how to automatically discover your PV inverter … 😬

wimleers, 2023-03-26:
… and of course just now I was able to _sort of_ reproduce this, but this seems to be just a timing problem:
```
[Live power flow@Power Monitor@Consumption] characteristic was supplied illegal value: number -2.9000000000000004 exceeded minimum of 0
[3/26/2023, 1:56:31 PM] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Consumption': characteristic was supplied illegal value: number -2.9000000000000004 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
```
I am already measuring how long it took to read the data from the inverter (`estimatedMsOffset`), I suspect that number will have been very high for this one erratic measurement.

What you're seeing is the same symptom but with a very different cause AFAICT.

wimleers, 2023-03-26:
Rescoped this to what I reported in my last comment; you're seeing this consistently due to #5.

wimleers, 2023-03-26:
🚢 Fix shipped in https://github.com/wimleers/homebridge-sma-home-manager/releases/tag/1.1.1

commit 75d3e905 (2023-03-26): Harden against the edge case where the inverter responds too slowly to have a valid measurement. Fixes #9.
closed 2023-03-26 by wimleers