wimleers/homebridge-sma-home-manager #12: [Feature Request] Please could you please include Values from the SMA Energy Meter for people without the Home Manager?

silvertristan, 2023-03-27:
I don't have a Home Manager as I don't really require it.

As the Old SMA plugin had I can see my current PV generation and Daily Yield. I would love to have the information like what my house is currently using, what's being fed into the grid plus daily yield and generation. I understand your plugin is to be able to control your homes energy usage using automations via the home app. For someone like me being able to see the value I have request is also a great help. Doing it all in the home app would be great rather than having to use the clunky SMA Energy app.

**Inverter:**

* **Model:** SB6.0-1AV-41 487

wimleers, 2023-03-27:
Based on the title, it sounds like you have an [SMA Energy Meter](https://www.sma.de/en/products/monitoring-control/sma-energy-meter) in addition to your inverter. Is that right?

If so: that _should_ work out of the box, at least according to SMA's own documentation at https://developer.sma.de/fileadmin/content/global/Partner/Documents/SMA_Labs/EMETER-Protokoll-TI-en-10.pdf.

But in your comment you only listed an inverter?

wimleers, 2023-03-27:
Could you post a photo of your SMA Energy Meter? 🙏

silvertristan, 2023-03-27:
[image]

wimleers, 2023-03-28:
That's odd, based on the photo and SMA's documentation, this should work just fine! 😬

Could you please:
- post the firmware version
- post a screenshot of your SMA Energy Manager's data appearing in Sunny Portal
- install https://apps.apple.com/be/app/discovery-dns-sd-browser/id305441017?l=nl and post a screenshot of what you see after tapping "World Wide Web HTTP"
- describe the network path between your SMA Energy Manager and device that Homebridge is running on (for example for me that's `SMA Energy Manager → switch → router → switch → Homebridge`)

wimleers, 2023-03-30:
Can you please update to https://github.com/wimleers/homebridge-sma-home-manager/releases/tag/1.1.3 and share the logged `DeviceType`? Thanks 🙏

silvertristan, 2023-03-31:
I'm at F1 in Melbourne all weekend. When I'm back on Monday I'll get on it. Thanks again.

silvertristan, 2023-04-11:
Hi,

Just to clarify I don't have an Energy Manager. I have a Energy Meter.

This is what I'm currently getting in the Homebridge log once updated:
}
[11/04/2023, 5:47:16 pm] [homebridge-sma-home-manager] Discovered SMA energy manager: no
[11/04/2023, 5:47:17 pm] [homebridge-sma-home-manager] Discovered SMA inverter: {
  DeviceClass: 'solar inverter',
  DeviceType: 9405,
  SerialNumber: 3007638487,
  FirmwareRevision: 'unknown'
}

Inverter Firmware version: 4.0.75.R
Devicetype: I can't see it anywhere or I don't know what you mean.
Network Path: SMA Energy Meter>(Ethernet)Sunnyboy Inverter>(WiFI)Router/Modem>(WiFi)Homebridge
Requested Screenshots below:
[image]
[image]
[image]

Thanks again for looking into this.
Cheers Tristan.

wimleers, 2023-05-28:
That all looks fine 😬 (Very similar to what I see!)

Your solar inverter is also of the exact same series as mine — you just have the 6 kW one, I have the 5 kW one.

Could you please:

1. Can you send me a screenshot of `https://169.254.12.3/#/devicemanagement` or `https://192.168.0.16/#/devicemanagement`? (You may need to log in first.)
2. describe the network path between your SMA Energy Manager and device that Homebridge is running on (for example for me that's SMA Energy Manager → switch → router → switch → Homebridge)

(I still suspect that this is a pure networking problem. i.e. that it's impossible for the computer on which you have Homebridge running to receive multicast packets from the Energy Meter.)

silvertristan, 2023-05-28:
[image]

Just to note and maybe it got missed in previous conversations just so we're not talking about two different things. I only have the inverter and energy meter. No energy manager. Hence why I just wanted to see valves and not controller the energy use like your plugin offers. With the old SMA plug-in I could see current generation and daily yield. With your plug-in you get a lot more info. I would also like to see current usage in addition to the generation and yield. 

If we can't get it worked out I don't want to waste your time. You've done a great job with this plug-in and I'm sure you've got bigger fish to fry. 

EnergyMeter(Ethernet)>Inverter(wifi)>router/modem(wifi)>homebridge(wifi)

Thanks again.

wimleers, 2023-05-28:
That all looks fine 👍 

>  I only have the inverter and energy meter. 
Yep!

>  Hence why I just wanted to see valves and not controller the energy use like your plugin offers.

Well that's the thing: the Home Manager and Energy Meter should be identical as far as this Homebridge plugin is concerned! The only difference is that the Home Manager can independently send signals to devices. For example, for me it controls my heatpump.
Either way, you should be able to get the same exact data, and create the same exact automations in Apple HomeKit.

> EnergyMeter(Ethernet)>Inverter(wifi)>router/modem(wifi)>homebridge(wifi)

Do you have a _single_ router/modem, no other WiFi access points?

silvertristan, 2023-05-28:
Just the one router/modem and that is the only access point. 

The modem does have 2 SSID's for 2.4Ghz and 5Ghz but both the inverter and homebridge are on 2.4.

wimleers, 2023-05-28:
That's for sure puzzling 🤔

It's a mystery to me why your homebridge instance wouldn't receive the multicast packets that your energy meter is broadcasting via IP multicast…

After some more pondering, I thought of one way this could still not work for you: is it possible that the device running Homebridge has:
- multiple connections to the the network? (one or more of Ethernet and/or one or more of WiFi)
- an IPv6 address, not IPv4?

If you're up for it: install Wireshark and observe! 😄

silvertristan, 2023-05-31:
Hey,

Checked it is IPv4 and no multiple connections to network. I'll have to install wireshark to see what's happening.

Since the 1.1.4 update I'm getting this:
`[31/05/2023, 10:22:41 am] [homebridge-sma-home-manager] Restarting Process...
[31/05/2023, 10:22:41 am] [homebridge-sma-home-manager] Launched child bridge with PID 7061
[5/31/2023, 10:22:41 AM] Registering platform 'homebridge-sma-home-manager.SMAHomeManager'
[31/05/2023, 10:22:41 am] [homebridge-sma-home-manager] Loaded homebridge-sma-home-manager v1.1.4 child bridge successfully

/var/lib/homebridge/node_modules/homebridge-sma-home-manager/index.js:45
        ['signals', 'surplusSignals'].forEach(requiredTopLevelKey => {
                               ^
Error: Invalid configuration: surplusSignals key is missing. See config.schema.json.
    at /var/lib/homebridge/node_modules/homebridge-sma-home-manager/index.js:47:10
    at Array.forEach (<anonymous>)
    at new SMAHomeManager (/var/lib/homebridge/node_modules/homebridge-sma-home-manager/index.js:45:32)
    at ChildBridgeFork.startBridge (/var/lib/homebridge/node_modules/homebridge/src/childBridgeFork.ts:150:42)
[31/05/2023, 10:22:41 am] [homebridge-sma-home-manager] Child bridge process ended
[31/05/2023, 10:22:41 am] [homebridge-sma-home-manager] Process Ended. Code: 1, Signal: null`

wimleers, 2023-06-01:
Okay. To clarify: if there were multiple connections, it would have been possible that this plugin was not correctly listening for multicast on _all_ network connections. But if there's only a single one, I don't see how that's possible.

See https://github.com/wimleers/homebridge-sma-home-manager/blob/5a3bd375e33f42165fe62bb1b79b21bcd486e001/index.js#L573-L620.

Actually, there's one thing you could try to do, change:
```js
this.socket.on('listening', function() {
			this.socket.addMembership(this.homeManagerAddress);
			this.multicastMembershipIntervalId = setInterval(this._keepMembershipActive.bind(this), 120*1000);
		}.bind(this));
```
to:
```js
this.socket.on('listening', function() {
			console.log('NOW LISTENING!');
			this.socket.addMembership(this.homeManagerAddress);
			this.multicastMembershipIntervalId = setInterval(this._keepMembershipActive.bind(this), 120*1000);
		}.bind(this));
```
to get a log entry to know for sure that the plugin actually started listening.

(But really, if it fails, then we should see Log entries generated by the `this.socket.on('error'` listener… 😬 )

P.S.: one thing I didn't ask you to try yet, is running Homebridge in debug mode. That will generate more log entries and _maybe_ that contains a helpful hint!

raimund-hdr, 2023-12-01:
If this issue is not resolved yet you may check the configuration of the energy meter. it is possible to change the multicast destinationthere . Maybe that has been changed. Sometimes installers change that to a unicast address of the inverter to overcome issues with powerline bridges which do not alway handle multicasts correctly (IGMP issue)