wimleers/homebridge-sma-home-manager #13: Support for SMA inverter Sunny Tripower Smart Energy

andrejditz, 2023-03-28:
I have a Sunny Tripower Smart Energy 10.0 and it looks like this type of inverter is not supported.
It is a hybrid inverter with a DC connection to a battery.

I got the following error in the Homebridge log:
The discovered inverter is not a solar inverter! Please create a bug report with as much detail as possible.

Please inform me if you need more information.

wimleers, 2023-03-28:
Oh, interesting! This is a _hybrid_ inverter AFAICT: solar & battery all in one?

Can you share your log output? 🙏

Scream4Life, 2023-03-29:
I get the exact error message when I hardcode my IP as recommended in version 1.1.1. It worked well with inserting it in 1.0.2. I think that has to do with querying the register codes which is new since 1.1.0.

https://github.com/wimleers/homebridge-sma-home-manager/issues/5#issuecomment-1484073248

wimleers, 2023-03-29:
@Scream4Life Yes, but you have 2 separate inverters, not a hybrid one. Different problem, same symptom!

andrejditz, 2023-03-29:
@wimleers yes that is the inverter i have in use.

Here is the log from the home bridge:

[29/03/2023, 09:22:26] [homebridge-sma-home-manager] Restarting child bridge...
[29/03/2023, 09:22:26] Got SIGTERM, shutting down child bridge process...
[29/03/2023, 09:22:27] [homebridge-sma-home-manager] Discovered SMA inverter: no
[29/03/2023, 09:22:27] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006XXXXXX, FirmwareRevision: '2.11.5.R' }
[29/03/2023, 09:22:28] [homebridge-sma-home-manager] Discovered SMA inverter: no
[29/03/2023, 09:22:28] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006XXXXXX, FirmwareRevision: '2.11.5.R' }
[29/03/2023, 09:22:29] [homebridge-sma-home-manager] Discovered SMA inverter: no
[29/03/2023, 09:22:29] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006XXXXXX, FirmwareRevision: '2.11.5.R' }
[29/03/2023, 09:22:30] [homebridge-sma-home-manager] Discovered SMA inverter: no
[29/03/2023, 09:22:30] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006XXXXXX, FirmwareRevision: '2.11.5.R' }
[29/03/2023, 09:22:31] [homebridge-sma-home-manager] Discovered SMA inverter: no
[29/03/2023, 09:22:31] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006XXXXXX, FirmwareRevision: '2.11.5.R' }
[29/03/2023, 09:22:31] [homebridge-sma-home-manager] Child bridge process ended
[29/03/2023, 09:22:31] [homebridge-sma-home-manager] Process Ended. Code: 143, Signal: null
[29/03/2023, 09:22:38] [homebridge-sma-home-manager] Restarting Process...
[29/03/2023, 09:22:39] [homebridge-sma-home-manager] Launched child bridge with PID 291867
[29/03/2023, 09:22:39] Registering platform 'homebridge-sma-home-manager.SMAHomeManager'
[29/03/2023, 09:22:39] [homebridge-sma-home-manager] Loaded homebridge-sma-home-manager v1.1.1 child bridge successfully
[29/03/2023, 09:22:39] Loaded 0 cached accessories from cachedAccessories.0E3D03ADB637.
[29/03/2023, 09:22:39] [homebridge-sma-home-manager] Found inverter SMA3012XXXXXX at SMA3012XXXXXX.local with IPv4 address 192.168.178.104.
[29/03/2023, 09:22:40] [homebridge-sma-home-manager] Discovered SMA inverter: no
[29/03/2023, 09:22:40] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006XXXXXX, FirmwareRevision: '2.11.5.R' }
[29/03/2023, 09:22:41] [homebridge-sma-home-manager] Discovered SMA inverter: {
  DeviceClass: 'unknown',
  SerialNumber: 3012XXXXXX,
  FirmwareRevision: 'unknown'
}
[29/03/2023, 09:22:41] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006XXXXXX, FirmwareRevision: '2.11.5.R' }
[29/03/2023, 09:22:41] [homebridge-sma-home-manager] The discovered inverter is not a solar inverter! Please create a bug report with as much detail as possible.
[29/03/2023, 09:22:59] [homebridge-sma-home-manager] This plugin is taking long time to load and preventing Homebridge from starting. See https://homebridge.io/w/JtMGR for more info.
[29/03/2023, 09:23:19] [homebridge-sma-home-manager] This plugin is taking long time to load and preventing Homebridge from starting. See https://homebridge.io/w/JtMGR for more info.
[29/03/2023, 09:23:39] [homebridge-sma-home-manager] This plugin is taking long time to load and preventing Homebridge from starting. See https://homebridge.io/w/JtMGR for more info.
[29/03/2023, 09:23:59] [homebridge-sma-home-manager] This plugin is taking long time to load and preventing Homebridge from starting. See https://homebridge.io/w/JtMGR for more info.
[29/03/2023, 09:24:19] [homebridge-sma-home-manager] This plugin is taking long time to load and preventing Homebridge from starting. See https://homebridge.io/w/JtMGR for more info.
[29/03/2023, 09:24:39] [homebridge-sma-home-manager] This plugin is taking long time to load and preventing Homebridge from starting. See https://homebridge.io/w/JtMGR for more info.
Error: Not running
    at new NodeError (node:internal/errors:399:5)
    at healthCheck (node:dgram:911:11)
    at Socket.dropMembership (node:dgram:859:3)
    at SMAHomeManager._keepMembershipActive (/var/lib/homebridge/node_modules/homebridge-sma-home-manager/index.js:801:15)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)
[29/03/2023, 09:24:39] [homebridge-sma-home-manager] Child bridge process ended
[29/03/2023, 09:24:39] [homebridge-sma-home-manager] Process Ended. Code: 1, Signal: null
[29/03/2023, 09:24:46] [homebridge-sma-home-manager] Restarting Process...
[29/03/2023, 09:24:47] [homebridge-sma-home-manager] Launched child bridge with PID 291881
[29/03/2023, 09:24:47] Registering platform 'homebridge-sma-home-manager.SMAHomeManager'
[29/03/2023, 09:24:47] [homebridge-sma-home-manager] Loaded homebridge-sma-home-manager v1.1.1 child bridge successfully
[29/03/2023, 09:24:47] Loaded 0 cached accessories from cachedAccessories.0E3D03ADB637.
[29/03/2023, 09:24:47] [homebridge-sma-home-manager] Found inverter SMA3012XXXXXX at SMA3012XXXXXX.local with IPv4 address 192.168.178.104.
[29/03/2023, 09:24:48] [homebridge-sma-home-manager] Discovered SMA inverter: no
[29/03/2023, 09:24:48] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006XXXXXX, FirmwareRevision: '2.11.5.R' }
[29/03/2023, 09:24:49] [homebridge-sma-home-manager] Discovered SMA inverter: {
[... cut ...]
[29/03/2023, 09:25:27] [homebridge-sma-home-manager] This plugin is taking long time to load and preventing Homebridge from starting. See https://homebridge.io/w/JtMGR for more info.
[29/03/2023, 09:25:47] [homebridge-sma-home-manager] This plugin is taking long time to load and preventing Homebridge from starting. See https://homebridge.io/w/JtMGR for more info.
[29/03/2023, 09:26:07] [homebridge-sma-home-manager] This plugin is taking long time to load and preventing Homebridge from starting. See https://homebridge.io/w/JtMGR f

Scream4Life, 2023-03-29:
> @Scream4Life Yes, but you have 2 separate inverters, not a hybrid one. Different problem, same symptom!

Yes, but when I hardcode the IP in the index.js it finds only the PV inverter and gives the message above. That’s why your current version does not work for my right now 😞

wimleers, 2023-03-29:
> > @Scream4Life Yes, but you have 2 separate inverters, not a hybrid one. Different problem, same symptom!
> 
> Yes, but when I hardcode the IP in the index.js it finds only the PV inverter and gives the message above. That’s why your current version does not work for my right now 😞

Yes. I am aware, as I wrote at https://github.com/wimleers/homebridge-sma-home-manager/issues/5#issuecomment-1484091901. Please stay tuned as I figure this out. I'm doing all this in my spare time! Blame SMA for A) making their devices work so inconsistently, B) not providing any documentation 😞

Thanks for your offer to schedule a remote session, I will take you up on your offer! 😊

wimleers, 2023-03-29:
@andrejditz Thanks, that looks like what I suspected! https://github.com/wimleers/homebridge-sma-home-manager/commit/f5db4fde5f52fbc67a312f953d2ecc834d9caace only looks at the device class. You're getting `unknown`, so it's _neither_ a `solar` _nor_ a `battery` inverter. That's _great_ news!

Expect an update in the coming days.

Scream4Life, 2023-03-29:
> > > @Scream4Life 
> Yes. I am aware, as I wrote at [#5 (comment)](https://github.com/wimleers/homebridge-sma-home-manager/issues/5#issuecomment-1484091901). Please stay tuned as I figure this out. I'm doing all this in my spare time! Blame SMA for A) making their devices work so inconsistently, B) not providing any documentation 😞
> 
> Thanks for your offer to schedule a remote session, I will take you up on your offer! 😊

Don‘t worry. I‘m thankful for every minute you‘ve spent with this. The solution is already great and I‘m sure it‘s getting even better over time. Next week I‘m off for holiday. Let me know when you‘ve time for a session afterwards.

wimleers, 2023-03-30:
I literally can't find any documentation detailing _all_ of the device classes … 😬 The closest I found was https://files.sma.de/downloads/ClusterController_MODBUS-TI-en-18.pdf, which lists:
```
Device class:
8000 = All devices
8001 = PV inverter
8002 = Wind power inverter
8007 = Battery inverter
8033 = Load
8064 = Sensor technology general
8065 = Energy meter
8128 = Communication products
```
So … I'll do a `1.1.2` release that hardcodes the knowledge of these classes and then adds better logging output so that @andrejditz can report back what he sees, so that I can _then_ do another release that fixes it for him 🤪

sbf-, 2023-03-30:
> I literally can't find any documentation detailing _all_ of the device classes

[Here they are](https://github.com/SBFspot/SBFspot/blob/master/SBFspot/TagListEN-US.txt#L4998)

```
8000=DevClss0\0\All Devices
8001=DevClss1\0\Solar Inverters
8002=DevClss2\0\Wind Turbine Inverter
8007=DevClss7\0\Battery Inverter
8008=DevClss8\0\Charging station
8009=DevClss9\0\Hybrid Inverter
8033=DevClss33\0\Load
8064=DevClss64\0\Sensor System in General
8065=DevClss65\0\Electricity meter
8066=DevClss66\0\Gas meter
8067=DevClss67\0\Generic meter
8096=DevClss96\0\Tracker
8128=DevClss128\0\Communication products
```

Except for 8009, there all coming from official SMA sources.
8009 was found by running SBFspot on a plant with a Hybrid Inverter

wimleers, 2023-03-30:
Done: https://github.com/wimleers/homebridge-sma-home-manager/commit/d7ff8ef49bd52d681b672d40a49cab744b20eabc

Shipped in https://github.com/wimleers/homebridge-sma-home-manager/releases/tag/1.1.2 — @andrejditz , please update to this version and report back what you're seeing in the log. That will give us the answers we need 😊

wimleers, 2023-03-30:
@sbf- HAH! I just shipped the logic that would have given me that same number. 😆 Well, let's hope that @andrejditz confirms it's `8009` then! 🤞 

It's interesting that you even have charging station, gas meter, generic meter and so on in there. Where did you find those?! 😳

P.S.: are you the `sbf` of https://github.com/SBFspot/SBFspot? 😮 You really seem to be 😅 If so: pleasure to meet you, and quite the honor to have you contributing here! 😊
P.P.S.: I've actually been looking into _stopping_ to use ModBus altogether and instead use Speedwire to read the data, that seems to be A) more reliable, B) more future-proof, C) how SMA devices actually exchange data between themselves. Unfortunately I've been unsuccessful at finding official documentation for this, and have only found partial implementations by others. If anybody is likely to know how to do this (or where to find docs for it), it's you, so … any chance you can point me in the right direction? 🤞

wimleers, 2023-03-30:
```
DeviceClass: 'unknown 8009',
```
That confirms it. Already committed that in https://github.com/wimleers/homebridge-sma-home-manager/commit/209ce84989bb18128c8c8011ab321f7f90f1d880.

Next up: explicitly allowing hybrid inverters 😊 Expect an update later today.

wimleers, 2023-03-30:
🚢 Shipped in https://github.com/wimleers/homebridge-sma-home-manager/releases/tag/1.1.3

closed 2023-03-30 by wimleers

sbf-, 2023-03-30:
> It's interesting that you even have charging station, gas meter, generic meter and so on in there. Where did you find those?

In the resource files of SunnyExplorer you can find a lot of info... Installed in `C:\Program Files (x86)\SMA\Sunny Explorer\resource\www_en.zip\TagList.All.TagListTitles.xml`

> I've actually been looking into _stopping_ to use ModBus altogether and instead use Speedwire to read the data
Modbus is the way to go... unlike SMAData2+ that is used with Speedwire it is well documented. SMAData2+ is a closed protocol, so there is no documentation available

andrejditz, 2023-03-30:
I installed the new version. It looks much better but still some problems.

Here again the log:
[30/03/2023, 14:16:42] [Homebridge UI] Homebridge log truncated by Homebridge.
[30/03/2023, 14:16:49] [homebridge-sma-home-manager] Inverter took 85.27356386184692 ms to respond, resulting in an invalid measurement. Dropping measurement.
[30/03/2023, 14:16:50] [homebridge-sma-home-manager] Inverter took 29.628110885620117 ms to respond, resulting in an invalid measurement. Dropping measurement.
[Live power flow@Power Monitor@Production] characteristic was supplied illegal value: number -120 exceeded minimum of 0
[30/03/2023, 14:16:56] [homebridge-sma-home-manager] This plugin generated a warning from the characteristic 'Production': characteristic was supplied illegal value: number -120 exceeded minimum of 0. See https://homebridge.io/w/JtMGR for more info.
[30/03/2023, 14:16:58] [homebridge-sma-home-manager] Inverter took 25.14814805984497 ms to respond, resulting in an invalid measurement. Dropping measurement.
[30/03/2023, 14:17:02] [homebridge-sma-home-manager] Inverter took 32.90649604797363 ms to respond, resulting in an invalid measurement. Dropping measurement.
[30/03/2023, 14:17:06] [homebridge-sma-home-manager] Inverter took 39.705671072006226 ms to respond, resulting in an invalid measurement. Dropping measurement.
[30/03/2023, 14:17:07] [homebridge-sma-home-manager] Inverter took 22.668567895889282 ms to respond, resulting in an invalid measurement. Dropping measurement.
[30/03/2023, 14:17:11] [homebridge-sma-home-manager] Inverter took 20.40967106819153 ms to respond, resulting in an invalid measurement. Dropping measurement.
[30/03/2023, 14:17:12] [homebridge-sma-home-manager] Inverter took 69.02382183074951 ms to respond, resulting in an invalid measurement. Dropping measurement.
[30/03/2023, 14:17:15] [homebridge-sma-home-manager] Inverter took 22.98326802253723 ms to respond, resulting in an invalid measurement. Dropping measurement.
[30/03/2023, 14:17:19] [homebridge-sma-home-manager] Inverter took 23.941329956054688 ms to respond, resulting in an invalid measurement. Dropping measurement.
[30/03/2023, 14:17:23] [homebridge-sma-home-manager] Inverter took 54.68346405029297 ms to respond, resulting in an invalid measurement. Dropping measurement.
[30/03/2023, 14:17:30] [homebridge-sma-home-manager] Inverter took 34.78252816200256 ms to respond, resulting in an invalid measurement. Dropping measurement.
[30/03/2023, 14:19:00] [homebridge-sma-home-manager] Inverter took 32.73114490509033 ms to respond, resulting in an invalid measurement. Dropping measurement.
[30/03/2023, 14:19:01] [homebridge-sma-home-manager] Inverter took 91.26431894302368 ms to respond, resulting in an invalid measurement. Dropping measurement.
[30/03/2023, 14:19:02] [homebridge-sma-home-manager] Inverter took 19.23366618156433 ms to respond, resulting in an invalid measurement. Dropping measurement.

Also the value in the production is not correct the SAM app is showing something with 4600W and the value from home bridge is 800w.

wimleers, 2023-03-30:
Could you please also provide the `DeviceType` that is being logged since `1.1.3`? 🙏

andrejditz, 2023-03-30:
The device type is 19051, see log below.

Here the log:

[30/03/2023, 16:16:30] [Homebridge UI] Homebridge log truncated by Homebridge.
[30/03/2023, 16:16:34] [homebridge-sma-home-manager] Restarting child bridge...
[30/03/2023, 16:16:34] Got SIGTERM, shutting down child bridge process...
[30/03/2023, 16:16:39] [homebridge-sma-home-manager] Child bridge process ended
[30/03/2023, 16:16:39] [homebridge-sma-home-manager] Process Ended. Code: 143, Signal: null
[30/03/2023, 16:16:46] [homebridge-sma-home-manager] Restarting Process...
[30/03/2023, 16:16:47] [homebridge-sma-home-manager] Launched child bridge with PID 319641
[30/03/2023, 16:16:47] Registering platform 'homebridge-sma-home-manager.SMAHomeManager'
[30/03/2023, 16:16:47] [homebridge-sma-home-manager] Loaded homebridge-sma-home-manager v1.1.3 child bridge successfully
[30/03/2023, 16:16:47] Loaded 0 cached accessories from cachedAccessories.0E3D03ADB637.
[30/03/2023, 16:16:48] [homebridge-sma-home-manager] Found inverter SMA3012416479 at SMA3012416479.local with IPv4 address 192.168.178.104.
[30/03/2023, 16:16:49] [homebridge-sma-home-manager] Discovered SMA inverter: no
[30/03/2023, 16:16:49] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006950550, FirmwareRevision: '2.11.5.R' }
[30/03/2023, 16:16:50] [homebridge-sma-home-manager] Discovered SMA inverter: {
  DeviceClass: 'hybrid inverter',
  DeviceType: 19051,
  SerialNumber: 3012416479,
  FirmwareRevision: 'unknown'
}
[30/03/2023, 16:16:50] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3006950550, FirmwareRevision: '2.11.5.R' }
[30/03/2023, 16:16:50] Initializing platform accessory 'Live power flow'...
[30/03/2023, 16:16:50] Initializing platform accessory 'Recent power flow'...
[30/03/2023, 16:16:50] Initializing platform accessory 'Today's energy flow'...
[30/03/2023, 16:16:50] Initializing platform accessory 'Off Grid energy signal'...
[30/03/2023, 16:16:50] Initializing platform accessory 'No Sun energy signal'...
[30/03/2023, 16:16:50] Initializing platform accessory 'High Import energy signal'...
[30/03/2023, 16:16:50] Homebridge v1.6.0 (HAP v0.11.0) (homebridge-sma-home-manager) is running on port 44138.
[30/03/2023, 16:19:00] [homebridge-sma-home-manager] Inverter took 47.93773794174194 ms to respond, resulting in an invalid measureme

OlsHub, 2023-04-05:
Hello, 

I have a similar issue with a Sunny Boy Storage 5.0:

[homebridge-sma-home-manager] The discovered inverter is not a solar or hybrid inverter! Please create a bug report with as much detail as possible. energyManager:
Model: 372,
SerialNumber: 3009878251, FirmwareRevision:
'2.11.5.R'
}, inverter: {
DeviceClass:
'battery inverter'
Devicelype: 9358,
SerialNumber: 3010761336, FirmwareRevision:
'unknown'

wimleers, 2023-04-07:
@OlsHub can you please install https://apps.apple.com/be/app/discovery-dns-sd-browser/id305441017?l=nl and make a screenshot of what you see after tapping “World Wide Web”? 🙏

OlsHub, 2023-04-07:
> @OlsHub can you please install https://apps.apple.com/be/app/discovery-dns-sd-browser/id305441017?l=nl and make a screenshot of what you see after tapping “World Wide Web”? 🙏

[image]

Website for SMA-Inverter: SMA3006893402._http._tcp.local.
World Wide Web HTTP

SMA3006893402.local:443
192.168.178.49:443

Website for SMA-Inverter: SMA3009496172-3._http._tcp.local.
World Wide Web HTTP

SMA3009496172-3.local:443
192.168.178.48:443

Website for SMA-Inverter: SMA3010761336._http._tcp.local.
World Wide Web HTTP

SMA3010761336.local:80
192.168.178.52:80

Website on SMA3009878251._http._tcp.local.
World Wide Web HTTP

SMA3009878251.local:80
192.168.178.4:80

I removed all IPV6 addresses... Seems not all of the inverters are offering https.

Thanks a lot for your help and for doing this!

commit 23bef97c (2023-03-30): Also allow inverters of the `hybrid inverter` device class. Fixes #13.