wimleers/homebridge-sma-home-manager #14: Does the plugin work with Modbus over TCP?

UffeBj, 2023-04-18:
**Describe Your Problem:**
I am trying to install the plugin on a Synology NAS.  I have activated Modbus on the SMA Inverter,Sunny boy 6.0 1AV-41, with the default Port 502, Unit ID 3 and the Modbus SunSpec profile version "Standard (recommended)". On the device configuration they talk about Modbus TCP or UDP settings (I have no Modbus knowledge). The error when trying to install refer to "modus-serial". Is there a fundamental miss here from my side trying to get it to work using Modbus TCP?

I have used the SMA's recommended tool for testing modbus connection and it works. https://www.sma-sunny.com/en/how-to-test-the-connection-to-your-sma-inverter/

**Paste of Logs:**

Here is the end of the log "npm ERR!     /var/packages/homebridge/home/.npm/_logs/2023-04-18T07_25_02_856Z-debug-0.log" as referred to when trying to install the plugin.

```
871 info run @serialport/bindings@8.0.8 install node_modules/homebridge-sma-home-manager/node_modules/@serialport/bindings prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
872 info run @serialport/bindings@8.0.8 install { code: 1, signal: null }
873 timing reify:rollback:createSparse Completed in 704ms
874 timing reify:rollback:retireShallow Completed in 0ms
875 timing command:install Completed in 25518ms
876 verbose stack Error: command failed
876 verbose stack     at ChildProcess.<anonymous> (/volume1/@appstore/homebridge/app/lib/node_modules/npm/node_modules/@npmcli/promise-spawn/lib/index.js:63:27)
876 verbose stack     at ChildProcess.emit (node:events:527:28)
876 verbose stack     at maybeClose (node:internal/child_process:1092:16)
876 verbose stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:302:5)
877 verbose pkgid @serialport/bindings@8.0.8
878 verbose cwd /volume1/homebridge
879 verbose Linux 3.10.108
880 verbose node v16.16.0
881 verbose npm  v8.11.0
882 error code 1
883 error path /volume1/homebridge/node_modules/homebridge-sma-home-manager/node_modules/@serialport/bindings
884 error command failed
885 error command sh -c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
886 verbose exit 1

```

**Config:**

```json

```

**Screenshots:**

**Inverter:**

* **Model:** Sunny boy 6.0 1AV-41,

**Homebridge Environment:**

* **Node.js Version**: v16.16.0
* **NPM Version**: 8.11.0
* **Homebridge Version**: v1.6.0
* **homebridge-sma-inverter Version**: v1.1.3 (2023-03-30)
* **Operating System**: Synology NAS, DSM 7.1.1-42962

wimleers, 2023-04-26:
Yes, it does. You're the first to report this. I wonder if this is related to the environment you're installing this in (Synology). I know of successful users on macOS and Raspberry Pi.

UffeBj, 2023-04-26:
Ok, thanks a lot. I will dig some more and see where I get.

wimleers, 2023-04-26:
1. this package depends on `modbus-serial`: https://www.npmjs.com/package/homebridge-sma-home-manager?activeTab=dependencies
2. `modbus-serial` depends on `serialport`: https://www.npmjs.com/package/modbus-serial?activeTab=dependencies

_That_ is the package that is failing to install.

wimleers, 2023-04-27:
It looks like there's quite a lot of installation problems for the `serialport` package, see https://github.com/serialport/node-serialport/issues?q=stack+Error%3A+command+failed.

Please open an issue there — not much I can do 😢

Good luck!

closed 2023-04-27 by wimleers

UffeBj, 2023-04-27:
Thanks a lot! Will give it a go next week.