wimleers/homebridge-sma-home-manager #16: [Feature Request] Move code to use SMA Speedwire Instead of Modbus

madtempest, 2023-07-29:
**Is your feature request related to a problem? Please describe:**
Hi,

My request is less about a problem and more about a potential problem: the security of Modbus on home networks. I can see that you have support for the SMA Speedwire protocol to allow it to work with SMA Home Manager and given the risks of enabling Modbus (SMA themselves recommend you don't enable it and it's not enabled by default for a reason), my request is to remove any code related to Modbus and use the Speedwire protocol instead.

This will improve the security of the plugin, the inverters and people's home networks.

**Describe the solution you'd like:**
Remove any code related to Modbus and use the Speedwire protocol instead.

**Describe alternatives you've considered:**
From a security perspective, there really isn't any alternatives The SMA Speedwire protocol supports encrypted communication whereas the Modbus protocol does not. Furthermore, there is a reduced risk of accidental harm or negative impact of local broadcast or other storm traffic on a network interacting with Speedwire vs Modbus.

sbf-, 2023-07-30:
Speedwire uses the closed and thus undocumented [SMAData2+ protocol](https://files.sma.de/downloads/Speedwire-TI-en-11.pdf)
Very unlikely Wim will put any further effort into Speedwire when there is such thing as Modbus

wimleers, 2023-08-02:
For now, I have to agree with @sbf-… as much as I would personally LOVE to move away from ModBus, the problem is indeed that the SpeedWire protocol is effectively undocumented.

Furthermore, there are significant caveats when enabling encryption for Speedwire in SMA's documentation. There's definitely no documentation for _encrypted_ Speedwire 😞 

I spent _dozens_ of hours trying to make that work. I found some hacky reverse-engineered sample implementations, but they're not guaranteed to keep working.

_If_ you can point me to official documentation for the (encrypted) Speedwire protocol, I'd be more than happy to change the implementation of this Homebridge plugin!

Unfortunately, SMA's actions suggest they would rather not have anybody interacting with their devices (despite the legal right to do so in the EU!) at all, let alone via widely adopted standards, or their own "standards" (which are entirely closed).

Sadly, ModBus is the only thing they explicitly support. And even then, their documentation often is not aligned with reality (see the numerous comments in `index.js` mentioning how the official docs indicate something ought to work but does not).

If anybody at SMA reads this: _please_ reach out to me via https://wimleers.com/contact, and I'd be more than happy to work with you! 😊 (And modify the contents of this comment.)

whistlefromthesun, 2024-01-08:
dunno where these come from - might be helpful?

[image]
[image]
[image]
[image]

wimleers, 2024-02-05:
@whistlefromthesun Can you please upload a PDF of that? These photos full of artifacts of a PDF on a computer screen (that almost looks like a CRT screen? 😅) are _very_ difficult to read 🤓

sbf-, 2024-02-05:
> @whistlefromthesun Can you please upload a PDF of that? These photos full of artifacts of a PDF on a computer screen (that almost looks like a CRT screen? 😅) are _very_ difficult to read 🤓

I guess these pictures were taken by a spy at SMA premises, knowing the fact SMAdata2+ protocol is undisclosed.
Interresting info for sure, if it´s readable.