wimleers/homebridge-sma-home-manager #18: [Support] Installation seems good, but no accessories appears

grrolland, 2023-10-07:
Installation seems good, but no accessories

I perform the plugin installation, configure Modbus exposition on the inverter, logs seems good but no accessories appear in the UI.

**Paste of Logs:**
```
[10/7/2023, 8:22:21 PM] [homebridge-sma-home-manager] Launched child bridge with PID 31603
[10/7/2023, 8:22:21 PM] Registering platform 'homebridge-sma-home-manager.SMAHomeManager'
[10/7/2023, 8:22:21 PM] [homebridge-sma-home-manager] Loaded homebridge-sma-home-manager v1.1.8 child bridge successfully
[10/7/2023, 8:22:21 PM] Loaded 0 cached accessories from cachedAccessories.0E498163E87C.
[10/7/2023, 8:22:21 PM] [homebridge-sma-home-manager] Found inverter SMA3008926192 at SMA3008926192.local with IPv4 address 10.0.4.54.
[10/7/2023, 8:22:22 PM] [homebridge-sma-home-manager] Discovered SMA inverter: no
[10/7/2023, 8:22:22 PM] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 349, SerialNumber: 1901470014, FirmwareRevision: '2.2.2.R' }
[10/7/2023, 8:22:23 PM] [homebridge-sma-home-manager] Discovered SMA inverter: {
  DeviceClass: 'solar inverter',
  DeviceType: 9405,
  SerialNumber: 3008926192,
  FirmwareRevision: 'unknown'
}
[10/7/2023, 8:22:23 PM] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 349, SerialNumber: 1901470014, FirmwareRevision: '2.2.2.R' }
[10/7/2023, 8:22:23 PM] Initializing platform accessory 'Live power flow'...
[10/7/2023, 8:22:23 PM] Initializing platform accessory 'Recent power flow'...
[10/7/2023, 8:22:23 PM] Initializing platform accessory 'Today's energy flow'...
[10/7/2023, 8:22:23 PM] Initializing platform accessory 'Off Grid energy signal'...
[10/7/2023, 8:22:23 PM] Initializing platform accessory 'No Sun energy signal'...
[10/7/2023, 8:22:23 PM] Homebridge v1.6.1 (HAP v0.11.1) (homebridge-sma-home-manager) is running on port 44037.
```

**Config:**
```json
{
    "signals": {
        "offGrid": true,
        "noSun": true,
        "highImport": false
    },
    "platform": "SMAHomeManager",
    "_bridge": {
        "username": "0E:49:81:63:E8:7C",
        "port": 44037
    }
}
```

**Inverter:**

* **Model:** Inverter Model : SUNNY BOY - SB6.01AV-41

**Homebridge Environment:**

* **Node.js Version**: v18.15.0
* **NPM Version**: unkown
* **Homebridge Version**: 1.6.1 (Synology Install)
* **homebridge-sma-inverter Version**: v1.1.8
* **Operating System**: Synology

grrolland, 2023-10-09:
Thanks for your answer.

I don't have Homekit (nor apple devices). I connect homebridge with node-red and [node-red-contrib-homebridge-automation](https://github.com/NorthernMan54/node-red-contrib-homebridge-automation)

I try to configure sma plugin in the parent bridge (uncheked the child bridge box) with same result => no accessories appear nor in node-red nor in homebridge.

wimleers, 2023-10-09:
What about other Homebridge plugins? Do their accessories appear?

This is a highly unusual setup, so good luck figuring this out 👍

grrolland, 2023-10-10:
Others plugins show accessories.

I try a different installation with the latest homebridge docker image, here is the conf : 

```
{
    "bridge": {
        "name": "Homebridge FEB1",
        "username": "0E:48:E5:73:FE:B1",
        "port": 51738,
        "pin": "280-69-275",
        "advertiser": "bonjour-hap"
    },
    "accessories": [],
    "platforms": [
        {
            "name": "Config",
            "port": 8581,
            "platform": "config"
        },
        {
            "signals": {
                "offGrid": true,
                "noSun": true,
                "highImport": false
            },
            "platform": "SMAHomeManager"
        }
    ]
}
```

This is the same behaviour. 

I turn the log in DEBUG mode. This is an extract : 

```
[10/10/2023, 9:31:00 AM] Loaded plugin: homebridge-sma-home-manager@1.1.8
[10/10/2023, 9:31:00 AM] Registering platform 'homebridge-sma-home-manager.SMAHomeManager'
[10/10/2023, 9:31:00 AM] ---
[10/10/2023, 9:31:00 AM] Loading 2 platforms...
[10/10/2023, 9:31:00 AM] [SMAHomeManager] Initializing SMAHomeManager platform...
[10/10/2023, 9:31:01 AM] [SMAHomeManager] Finding inverter using mDNS/DNS-SD…
[10/10/2023, 9:31:01 AM] [SMAHomeManager] Found inverter DNS-SD service: {
  addresses: [ 'fe80::240:adff:febf:138', '10.0.4.54' ],
  subtypes: [],
  rawTxt: [ <Buffer > ],
  txt: {},
  name: 'Website for SMA-Inverter: SMA3008926192',
  fqdn: 'Website for SMA-Inverter: SMA3008926192._http._tcp.local',
  host: 'SMA3008926192.local',
  referer: { address: '10.0.4.54', family: 'IPv4', port: 5353, size: 173 },
  port: 443,
  type: 'http',
  protocol: 'tcp'
}
[10/10/2023, 9:31:01 AM] [SMAHomeManager] Found inverter SMA3008926192 at SMA3008926192.local with IPv4 address 10.0.4.54.
[10/10/2023, 9:31:01 AM] [SMAHomeManager] Connecting to inverter at 10.0.4.54.
2023-10-10T07:31:01.183Z modbus-serial TCP port: signal connect
[10/10/2023, 9:31:01 AM] [SMAHomeManager] Successfully connected to inverter.
2023-10-10T07:31:02.051Z modbus-serial {
  action: 'send tcp port',
  data: <Buffer 03 03 75 63 00 04 af f9>,
  buffer: <Buffer 00 01 00 00 00 06 03 03 75 63 00 04>,
  unitid: 3,
  functionCode: 3,
  transactionsId: 1
}
2023-10-10T07:31:02.053Z modbus-serial {
  action: 'send tcp port',
[... 142 lines cut ...]
  buffer: <Buffer 00 06 00 00 00 06 03 03 78 37 00 02>,
  unitid: 3,
  functionCode: 3,
  transactionsId: 6
}
2023-10-10T07:31:05.286Z modbus-serial {
  action: 'receive tcp port strings',
  data: <Buffer 00 06 00 00 00 07 03 03 04 00 00 03 60>
}
2023-10-10T07:31:05.288Z modbus-serial {
  action: 'parsed tcp port',
  buffer: <Buffer 03 03 04 00 00 03 60 d9 2b>,
  transactionId: 6
}
```

And always no accessories :-( : 
[image]

And nothing in node-red ...

[image]

But with the previous homebridge instance, there are some accessories in node-red : 
[image]

wimleers, 2023-10-27:
You can see that the accessories were initialized:
```
[10/10/2023, 9:31:03 AM] Initializing platform accessory 'Live power flow'...
[10/10/2023, 9:31:03 AM] Initializing platform accessory 'Recent power flow'...
[10/10/2023, 9:31:03 AM] Initializing platform accessory 'Today's energy flow'...
[10/10/2023, 9:31:03 AM] Initializing platform accessory 'Off Grid energy signal'...
[10/10/2023, 9:31:03 AM] Initializing platform accessory 'No Sun energy signal'...
```
**⚠️ Note: That is a message from `homebridge` itself, not my plugin!**

Perhaps it's because these accessories are of a _custom type_, and that's why it doesn't work? These don't work in Apple Home either (you _see_ them but they are labeled "Unsuported").

See https://github.com/wimleers/homebridge-sma-home-manager/blob/1.1.8/index.js#L274-L286 and https://github.com/wimleers/homebridge-sma-home-manager/blob/1.1.8/index.js#L177-L211.

Please create an issue against https://github.com/NorthernMan54/node-red-contrib-homebridge-automation to get help from that side 😊

wimleers, 2023-10-27:
A quick search makes me 99% confident that this is the reason.

See https://github.com/NorthernMan54/node-red-contrib-homebridge-automation/blob/77e0343ecc3d3dbb457636d23a8df6b105538a24/lib/Service.js.

It ends with:
```
    default:
      debug("Missing HB Type", id);
  }
```
I bet you'll see that in your debug logs for _that_ piece of software 👍