wimleers/homebridge-sma-home-manager #19: SunnyBoy SB2.5 not discovered

raimund-hdr, 2023-11-17:
**The "old" Sunnyboy 2.5 will not respond to DNS/SD:**

My "old SB2.5 with updated firmware is not discovered. I can find him with his SMAxxxxxxx.local  Address but he will not be found.

It'd be great to add "manual" configuration for the inverter.

**Paste of Logs:**

[17.11.2023, 09:28:30] [homebridge-sma-home-manager] Discovered SMA inverter: no
[17.11.2023, 09:28:30] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 349, SerialNumber: 1901400245, FirmwareRevision: '2.0.16.R' }
[17.11.2023, 09:28:31] [homebridge-sma-home-manager] Discovered SMA inverter: no
[17.11.2023, 09:28:31] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 349, SerialNumber: 1901400245, FirmwareRevision: '2.0.16.R' }
[17.11.2023, 09:28:32] [homebridge-sma-home-manager] Discovered SMA inverter: no
[17.11.2023, 09:28:32] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 349, SerialNumber: 1901400245, FirmwareRevision: '2.0.16.R' }
[17.11.2023, 09:28:33] [homebridge-sma-home-manager] Discovered SMA inverter: no
[17.11.2023, 09:28:33] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 349, SerialNumber: 1901400245, FirmwareRevision: '2.0.16.R' }
[17.11.2023, 09:28:34] [homebridge-sma-home-manager] Discovered SMA inverter: no
[17.11.2023, 09:28:34] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 349, SerialNumber: 1901400245, FirmwareRevision: '2.0.16.R' }
[17.11.2023, 09:28:35] [homebridge-sma-home-manager] Discovered SMA inverter: no
[17.11.2023, 09:28:35] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 349, SerialNumber: 1901400245, FirmwareRevision: '2.0.16.R' }
[17.11.2023, 09:28:35] [homebridge-sma-home-manager] This plugin is taking long time to load and preventing Homebridge from starting. See https://homebridge.io/w/JtMGR for more info.
[17.11.2023, 09:28:36] [homebridge-sma-home-manager] Discovered SMA inverter: no
[17.11.2023, 09:28:36] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 349, SerialNumber: 1901400245, FirmwareRevision: '2.0.16.R' }

**Inverter:**

* **Model:** SunnyBoy SB2.5

**Konfiguration**
* **OS:** Ubuntu Jammy Jellyfish (22.04.3 LTS)
* **homebridge:** v1.7.0
* **Node.js Version:** v18.16.0

raimund-hdr, 2023-11-28:
Additional question: 
Do you use the official device discovery? 
Drom "https://developer.sma.de/on-site-integration.html":

**SPEEDWIRE DEVICE DISCOVERY protocol interface**
With Speedwire Device Discover, devices in the local network can be detected.
[for all SMA Speedwire or SMA Speedwire/Webconnect devices](https://developer.sma.de/fileadmin/content/global/Partner/Documents/sma_developer/SpeedwireDD-TI-en-10.pdf)
[SpeedwireDD-TI-en-10.pdf](https://github.com/wimleers/homebridge-sma-home-manager/files/13488956/SpeedwireDD-TI-en-10.pdf)

wimleers, 2023-11-29:
What do you see if you run `dns-sd -B` on a macOS computer, or use the "Discovery" app on iOS?

raimund-hdr, 2023-11-29:
btw (additional information to my previous mail)
The SMA device discovered is a SMA EV-Charger ;-)

wimleers, 2023-12-07:
> since the SunnyBoy 2.5 and other elder models do not have OS on board they do not answer to service discovery nor have a web server available.

Woah! That must be _very_ old 😄 

> There is an original SMA documentation on how to discover SMA inverters in a network which should work for ALL inverter. I attach the document to this mail, it includes information on how to proceed as well as some sample code.

I can't see that document — GitHub's "reply to e-mail to post on the issue" service is 💩 🙈 

WRT Speedwire: see #16 — even the maintainer of https://github.com/SBFspot/SBFspot says there's no docs for this 😅 

So … if you have such a document, _please_ do upload it here, on the GitHub issue! 🙏 🤩 

> The SMA device discovered is a SMA EV-Charger ;-)

Aha! 😅 I _was_ going to ask about that. Thanks!

wimleers, 2023-12-07:
WRT SpeedWire being closed source, there's also #4. See https://github.com/wimleers/homebridge-sma-home-manager/issues/4#issuecomment-1468669369. A firmware update of the inverters caused new messages to appear, which are completely undocumented.

This was also reported against other projects, such as https://github.com/SBFspot/SBFspot/issues/626.

whistlefromthesun, 2023-12-11:
OK - sorry for being afk for some time.
I already attached the SMA documentation (original from SMA Developer Website) above in this thread. But here ist again:

[SpeedwireDD-TI-en-10.pdf](https://github.com/wimleers/homebridge-sma-home-manager/files/13633560/SpeedwireDD-TI-en-10.pdf)

I hope this can help - discovery will work for ALL SMA devices (starting from around 2011).

sbf-, 2023-12-11:
> [SpeedwireDD-TI-en-10.pdf](https://github.com/wimleers/homebridge-sma-home-manager/files/13633560/SpeedwireDD-TI-en-10.pdf)
> 
> I hope this can help - discovery will work for ALL SMA devices (starting from around 2011).

Be aware the Python example in the document has an error:
`spwMCastAdr = 239.12.255.255`
should be
`spwMCastAdr = 239.12.255.254`

wimleers, 2023-12-18:
Thanks for that, both of you!

@sbf- , are you using Speedwire discovery as described in that document in https://github.com/SBFspot/SBFspot?

sbf-, 2023-12-18:
> @sbf- , are you using Speedwire discovery as described in that document in https://github.com/SBFspot/SBFspot?

Yes, when IP=0.0.0.0 is configured
[Speedwire discovery in SBFspot](https://github.com/SBFspot/SBFspot/blob/V3.9.7/SBFspot/SBFspot.cpp#L320)

raimund-hdr, 2024-02-16:
Any additional information on that issue? 
I think this is linked to issue: "Inverter not discovered":
https://github.com/wimleers/homebridge-sma-home-manager/issues/20

wimleers, 2026-10-15:
> [!NOTE]
> This comment was AI-generated, using Claude Code.

@raimund-hdr You were right: SMA *does* document a discovery mechanism that works for inverters that don't advertise themselves via mDNS/DNS-SD! Thanks for pointing to it, and sorry it took this long. 😅

Fixed in #25: the next release will discover inverters using SMA's [Speedwire Device Discovery](https://developer.sma.de/fileadmin/content/www.developer.sma.de/docs/SpeedwireDD-TI-en-10.pdf) as well, next to the existing mDNS/DNS-SD discovery (unchanged). Every SMA device answers it, so each one is checked over ModBus, and only solar and hybrid inverters are accepted. It retries every 30 seconds until an inverter is found, because inverters switch off at night.

That should make manual configuration unnecessary.

I could only verify this on my own inverter (a Sunny Boy Smart Energy 3.6), not on a Sunny Boy 2.5. It should work as long as:
- the inverter's "Modbus TCP server" is enabled (a firmware update can reset it)
- Homebridge runs in the same network as the inverter, so that multicast reaches it

Please try the next release and let me know! If it still says `Discovered SMA inverter: no`, please enable debug logging in Homebridge and comment here with the lines that mention `Speedwire discovery`. 🙏

closed 2026-10-15 by wimleers