wimleers/homebridge-sma-home-manager #21: [Bug] Bridge restarting, SHM, SBS, Inverter not available

funx1, 2024-02-12:
**Paste of Logs:**
```
[12/02/2024, 11:01:10] [homebridge-sma-home-manager] Restarting Process...
[12/02/2024, 11:01:12] [homebridge-sma-home-manager] Launched child bridge with PID 20923
[12/02/2024, 11:01:13] Registering platform 'homebridge-sma-home-manager.SMAHomeManager'
[12/02/2024, 11:01:13] [homebridge-sma-home-manager] Loaded homebridge-sma-home-manager v1.1.8 child bridge successfully
[12/02/2024, 11:01:13] Loaded 0 cached accessories from cachedAccessories.0E925CDAD228.
[12/02/2024, 11:01:13] [homebridge-sma-home-manager] Found inverter SMA30120xxxx at SMA3012xxx.local with IPv4 address 192.168.179.182.
[12/02/2024, 11:01:14] [homebridge-sma-home-manager] Discovered SMA inverter: no
[12/02/2024, 11:01:14] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 30118xxxx, FirmwareRevision: '2.13.6.R' }
/var/lib/homebridge/node_modules/homebridge-sma-home-manager/index.js:402
                                const deviceClassId = data.buffer.slice(0, 4).readUInt32BE();
                               ^
TypeError: Cannot read properties of undefined (reading 'buffer')
    at SMAHomeManager.<anonymous> (/var/lib/homebridge/node_modules/homebridge-sma-home-manager/index.js:402:32)
    at TcpPort.<anonymous> (/var/lib/homebridge/node_modules/homebridge-sma-home-manager/node_modules/modbus-serial/index.js:340:37)
    at TcpPort.emit (node:events:517:28)
    at Socket.<anonymous> (/var/lib/homebridge/node_modules/homebridge-sma-home-manager/node_modules/modbus-serial/ports/tcpport.js:77:20)
    at Socket.emit (node:events:517:28)
    at addChunk (node:internal/streams/readable:368:12)
    at readableAddChunk (node:internal/streams/readable:341:9)
    at Socket.Readable.push (node:internal/streams/readable:278:10)
    at TCP.onStreamRead (node:internal/stream_base_commons:190:23)
[12/02/2024, 11:01:14] [homebridge-sma-home-manager] Child bridge process ended
[12/02/2024, 11:01:14] [homebridge-sma-home-manager] Process Ended. Code: 1, Signal: null
[12/02/2024, 11:01:18] [homebridge-sma-home-manager] Restarting child bridge...
[12/02/2024, 11:01:21] [homebridge-sma-home-manager] Restarting Process...
[12/02/2024, 11:01:23] [homebridge-sma-home-manager] Launched child bridge with PID 20937
[12/02/2024, 11:01:23] Registering platform 'homebridge-sma-home-manager.SMAHomeManager'
[12/02/2024, 11:01:24] [homebridge-sma-home-manager] Loaded homebridge-sma-home-manager v1.1.8 child bridge successfully
[12/02/2024, 11:01:24] Loaded 0 cached accessories from cachedAccessories.0E925CDAD228.
[12/02/2024, 11:01:24] [homebridge-sma-home-manager] Found inverter SMA30120xxxxx at SMA30120xxxxx.local with IPv4 address 192.168.179.182.
[12/02/2024, 11:01:25] [homebridge-sma-home-manager] Discovered SMA inverter: no
[12/02/2024, 11:01:25] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 301184xxxx, FirmwareRevision: '2.13.6.R' }
[12/02/2024, 11:01:25] [homebridge-sma-home-manager] Child bridge process ended
[12/02/2024, 11:01:25] [homebridge-sma-home-manager] Process Ended. Code: 1, Signal: null
[12/02/2024, 11:01:32] [homebridge-sma-home-manager] Restarting Process...
[12/02/2024, 11:01:34] [homebridge-sma-home-manager] Launched child bridge with PID 20951
[12/02/2024, 11:01:34] [homebridge-sma-home-manager] Loaded homebridge-sma-home-manager v1.1.8 child bridge successfully
[12/02/2024, 11:01:35] [homebridge-sma-home-manager] Found inverter SMA30120xxxxx at SMA3012001984.local with IPv4 address 192.168.179.182.
[12/02/2024, 11:01:36] [homebridge-sma-home-manager] Discovered SMA inverter: no
[12/02/2024, 11:01:36] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 301184xxxx, FirmwareRevision: '2.13.6.R' }
[12/02/2024, 11:01:36] [homebridge-sma-home-manager] Child bridge process ended
[12/02/2024, 11:01:36] [homebridge-sma-home-manager] Process Ended. Code: 1, Signal: null
[12/02/2024, 11:01:43] [homebridge-sma-home-manager] Restarting Process...
[12/02/2024, 11:01:45] [homebridge-sma-home-manager] Launched child bridge with PID 20965
[12/02/2024, 11:01:45] [homebridge-sma-home-manager] Loaded homebridge-sma-home-manager v1.1.8 child bridge successfully
[12/02/2024, 11:01:46] [homebridge-sma-home-manager] Found inverter SMA30120xxxxx at SMA30120xxxx.local with IPv4 address 192.168.179.182.
[12/02/2024, 11:01:47] [homebridge-sma-home-manager] Discovered SMA inverter: no
[12/02/2024, 11:01:47] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 30118xxxx, FirmwareRevision: '2.13.6.R' }```

**Config:**
<!-- Paste relevant output between the two ``` lines below -->
<!-- Remove any sensitive information, passwords, etc. -->

```json

```

**Screenshots:**

**Inverter:** SHM

* **Model:** 2.0

**Homebridge Environment:**
local network

* **Node.js Version**:  v18.19.0
* **NPM Version**:  not installed
* **Homebridge Version**:Linux homebridge 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux
* **homebridge-sma-inverter Version**: v1.1.8
* **Operating System**: Raspbian

funx1, 2024-02-26:
Some ideas?

wimleers, 2024-03-02:
```
/var/lib/homebridge/node_modules/homebridge-sma-home-manager/index.js:402
                                const deviceClassId = data.buffer.slice(0, 4).readUInt32BE();
                               ^
TypeError: Cannot read properties of undefined (reading 'buffer')
```
☝️ Looking at the relevant code:
```
			// Read device class (U32, ENUM).
			client.readHoldingRegisters(30051, 4, function(err, data) {
				const deviceClassId = data.buffer.slice(0, 4).readUInt32BE();
				deviceType = data.buffer.slice(4, 8).readUInt32BE();
				deviceClass = SMA_DEVICE_CLASSES.get(deviceClassId) || `unknown ${deviceClassId}`;
				registerInverterDiscoveryIfComplete();
			}.bind(this));
```

That's blindly reading `data` and is ignoring `err` altogether 😬 This is one of the last remaining patterns of code remaining of the original codebase that I forked from (https://github.com/codyc1515/homebridge-sma-inverter).

**Question: Modbus TCP/IP is enabled?**

funx1, 2024-03-08:
TCP is active on the STP and SBS. Actually I can access the SBS from the HA SMA implementation but not the STP. But I was able to access values over iobroker and modus.

wimleers, 2024-03-11:
> Actually I can access the SBS from the HA SMA implementation

Last time I looked, the Home Assistant integration was _scraping the inverter's web UI_ 😨

> But I was able to access values over iobroker and modus.

How does iobroker get the data? And what is "modus"?

wimleers, 2026-10-15:
> [!NOTE]
> This comment was AI-generated, using Claude Code.

@funx1 Sorry this took so long! Your log shows what happened:

1. `Found inverter SMA30120xxxx … with IPv4 address 192.168.179.182`: mDNS/DNS-SD found an SMA device. In 1.1.8, the plugin trusted that blindly.
2. It then tried to read that device's device class (register 30051) over ModBus. That read failed. The plugin ignored the error and accessed the missing data anyway: `TypeError: Cannot read properties of undefined (reading 'buffer')`.
3. That crashed the child bridge, Homebridge restarted it, and the same thing happened again: a crash loop.

The next release (1.2.0) changes two things:
- Every device found via mDNS/DNS-SD is first verified over ModBus, in a way that catches errors. A device whose device class cannot be read, or that is not a solar or hybrid inverter (e.g. your Sunny Boy Storage, which is a battery inverter), is ignored, so no crash. (6626b43381403482ed69875c8d89fad883b1d7d0)
- Inverters are now also discovered via SMA's Speedwire discovery. So your STP 15000TL should be found that way too, as long as its "Modbus TCP server" is enabled. (8bd9d9169fd25a33f70121ce1937ae1a089455ca, see #25)

This issue stays open for the remaining work: other ModBus reads still ignore errors the same way, so a ModBus failure *after* discovery can still crash the plugin. That is planned for v2.

If you ever give the plugin another try: I'd love to hear whether 1.2.0 finds your STP 15000TL!