wimleers/homebridge-sma-home-manager #24: [Bug] On Install on Synology

Arthuro87, 2024-03-27:
Please make it run on a Synology DSM. On Synology you can run the Package Homebridge. 

Running post-install scripts...

> @serialport/bindings@8.0.8 install
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

gyp ERR! build error 
gyp ERR! stack Error: not found: make
gyp ERR! stack at getNotFoundError (/volume1/@appstore/homebridge/app/lib/node_modules/npm/node_modules/which/lib/index.js:16:17)
gyp ERR! stack at which (/volume1/@appstore/homebridge/app/lib/node_modules/npm/node_modules/which/lib/index.js:77:9)
gyp ERR! stack at async doWhich (/volume1/@appstore/homebridge/app/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:112:22)
gyp ERR! stack at async loadConfigGypi (/volume1/@appstore/homebridge/app/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:77:7)
gyp ERR! stack at async build (/volume1/@appstore/homebridge/app/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:35:3)
gyp ERR! stack at async run (/volume1/@appstore/homebridge/app/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js:81:18)
gyp ERR! System Linux 4.4.302+
gyp ERR! command "/volume1/@appstore/Node.js_v20/usr/local/bin/node" "/volume1/@appstore/homebridge/app/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /volume1/homebridge/node_modules/homebridge-sma-home-manager/node_modules/@serialport/bindings
gyp ERR! node -v v20.9.0
gyp ERR! node-gyp -v v10.0.1
gyp ERR! not ok 
npm ERR! code 1
npm ERR! path /volume1/homebridge/node_modules/homebridge-sma-home-manager/node_modules/@serialport/bindings
npm ERR! command failed
npm ERR! command sh -c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

SpaceDragon1205, 2024-04-07:
Same at me. Using a Synology DS720+ and it's not possible to install this Plugin. Same failure.

wimleers, 2026-10-15:
> [!NOTE]
> This comment was AI-generated, using Claude Code.

@Arthuro87 @SpaceDragon1205 Found the root cause:

1. This plugin depends on `modbus-serial` `^7.7.3`, which resolves to 7.8.1. That version *requires* `serialport` 8, which needs the native module `@serialport/bindings@8.0.8`.
2. `prebuild-install` first tries to download a prebuilt binary for it. But [`@serialport/bindings@8.0.8`](https://github.com/serialport/node-serialport/releases/tag/%40serialport%2Fbindings%408.0.8) dates from May 2020: it only has prebuilt binaries up to Node 12, and you're running Node 20.
3. So npm falls back to building it from source (`node-gyp rebuild`). That requires build tools like `make`, which Synology DSM does not have, so the installation fails.

The irony: this plugin only talks ModBus over TCP, and never uses a serial port. 🙈

The fix: since `modbus-serial` 8.0.21, `serialport` is an *optional* dependency (https://github.com/yaacov/node-modbus-serial/pull/590). npm continues installing when an optional dependency fails to build. So requiring `modbus-serial` `^8.0.21` fixes this, without needing any build tools. That's planned for v2.

#14 had the exact same cause.