wimleers/homebridge-sma-home-manager #25: [Bug] Discovered SMA inverter: no

Stefan-83, 2024-06-20:
Hi, 
the plugin shows me the message "Discovered SMA inverter: no"
Modbus is activated:
[image]

**Paste of Logs:**
```
[20/06/2024, 11:16:07] [homebridge-sma-home-manager] Launched child bridge with PID 1163840
[20/06/2024, 11:16:07] [homebridge-sma-home-manager] Loaded homebridge-sma-home-manager v1.1.8 child bridge successfully
[20/06/2024, 11:16:08] [homebridge-sma-home-manager] Discovered SMA inverter: no
[20/06/2024, 11:16:08] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3002xxxxx, FirmwareRevision: '2.14.13.R' }
[20/06/2024, 11:16:09] [homebridge-sma-home-manager] Discovered SMA inverter: no
[20/06/2024, 11:16:09] [homebridge-sma-home-manager] Discovered SMA energy manager: { Model: 372, SerialNumber: 3002xxxxx, FirmwareRevision: '2.14.13.R' }

```

**Config:**
{
    "signals": {
        "offGrid": true,
        "noSun": true,
        "highImport": true
    },
    "platform": "SMAHomeManager",
    "_bridge": {
        "username": "xxxxxxxxxx",
        "port": 31792
    }
}

```json

```

**Inverter:**  
--

* **Model:** Sunny Tripower 10000TL-20

**Homebridge Environment:**

* **Node.js Version**: v20.12.2
* **Homebridge Version**: 1.8.3
* **homebridge-sma-inverter Version**: v1.1.8
* **Operating System**: Raspbian

wimleers, 2025-05-18:
I'm having the same problem after my SMA Sunny Boy 5.0 (PV inverter) got swapped for a SMA Sunny Boy Smart Energy 3.0 (hybrid inverter).

It's a mDNS/Bonjour problem, because https://github.com/onlxltd/bonjour-service doesn't find it, yet `dns-sd -B` _does_! 😬

stha, 2025-06-30:
@wimleers Any update here? I've got the same problem with my Sunny Tripower 20000TL-30. Modbus TCP Server is enabled, but the inverter doesn't get discovered by the homebridge plugin.

wimleers, 2026-10-15:
> [!NOTE]
> This comment was AI-generated, using Claude Code.

Found the root cause for my Sunny Boy Smart Energy 3.6 (SBSE3.6-50)! 🥳 It has two causes:
1. It advertises `Website for SMA-Product: SMA<serial>` instead of `Website for SMA-Inverter`, which the plugin required.
2. It sends only a PTR record, without SRV/A records in the same packet. That's why `dns-sd -B` sees it but the plugin does not: https://github.com/onlxltd/bonjour-service never reports such a service.

https://github.com/onlxltd/bonjour-service/blob/74aa9de5e896cb185ba69711ab0288219cd1adfe/src/lib/browser.ts#L216-L237

Fixed in https://github.com/wimleers/homebridge-sma-home-manager/commit/8bd9d9169fd25a33f70121ce1937ae1a089455ca: the next release will discover inverters using SMA's documented [Speedwire Device Discovery](https://developer.sma.de/fileadmin/content/www.developer.sma.de/docs/SpeedwireDD-TI-en-10.pdf) as well, next to the existing mDNS/DNS-SD discovery (unchanged). Every SMA device answers it, so each one is checked over ModBus, and only solar and hybrid inverters are accepted. It retries every 30 seconds until an inverter is found, because inverters switch off at night.

@Stefan-83 @stha I could only verify this on my SBSE. I can't say for sure that it also fixes your STP 10000TL-20 and STP 20000TL-30, but it should, as long as:
- the inverter's "Modbus TCP server" is enabled (a firmware update can reset it)
- Homebridge runs in the same network as the inverter, so that multicast reaches it

Please try the next release and let me know! If it still says `Discovered SMA inverter: no`, enable debug logging in Homebridge and share the lines that mention `Speedwire discovery`. 🙏

closed 2026-10-15 by wimleers